nanopb/nanopb #935: int_size IS_8 takes two bytes for each uint8

hpsaturn, 2024-01-29:
## Overview

I don't know yet if this is an issue. I'm trying to reduce the current size of my payload or encoded stream, and I notice that for each insertion of each uint8, the stream add two bytes, and should be one byte, I guess. I tested different options but nothing. 

My proto definition:

```cpp
syntax = "proto2";

import "nanopb.proto";

message Frame {
  required uint32 lenght = 1 [(nanopb).int_size = IS_16];
  repeated uint32 data = 2 [(nanopb).int_size = IS_8, (nanopb).max_size = 1];
}
```

My code:

```cpp
bool encode_uint8_array(pb_ostream_t *stream, const pb_field_t *field, void *const *arg) {
  for (int i = 0; i < chunk_size; i++) {
    if (!pb_encode_tag_for_field(stream, field))
      return false;
    uint8_t val = (out_jpg+chunk_pos)[i];
    if (!pb_encode_varint(stream, val))
      return false;
  }
  return true;
}
```

[Full code](https://github.com/hpsaturn/esp32s3-cam/blob/master/examples/freenove-espnow/freenove-espnow.cpp)

PetteriAimonen, 2024-01-29:
Varints with value >= 128 use two bytes when encoded: https://protobuf.dev/programming-guides/encoding/#varints

For data like that, you probably want to use the `bytes` type in protobuf, which doesn't use extra space and automatically passes the length too.

hpsaturn, 2024-01-29:
Thanks, I didn't know that, and I tried with bytes but I don't remember why it didn't work in the beginning of this exploration. Also with `repeated bytes` I had issues.. Maybe do you know an example or test using `bytes` and callback similar a `repeated int` that I can follow? Thanks

PetteriAimonen, 2024-01-29:
Encoding bytes is the same as encoding string here, just replace strlen() with the actual binary data length:
https://github.com/nanopb/nanopb/blob/master/tests/callbacks/encode_callbacks.c#L9-L17

hpsaturn, 2024-01-29:
Thanks, I did a mistake the first time that I try with encoding string, I was close. Thanks for your fast feedback and for this wonderful library. Is working better:

https://github.com/nanopb/nanopb/assets/423856/31a478ff-f916-4ad5-97eb-20238359e0eb

closed 2024-07-19 by PetteriAimonen