nanopb/nanopb #937: nanopb_generator.py doesn't recognize pb_byte_t in .proto files

shersey-locus, 2024-02-06:
When I write a .proto file and tag a field as `pb_byte_t`, the generator in nanopb-0.4.7 gives me an error that `"pb_byte_t" is not defined.` I can see definitions of pb_byte_t in C header files, but the generator isn't having any.

If I use uint32 as the field type, everything works.

Command line is here:
```
$ $PYTHONPATH/python libraries/nanopb/generator/nanopb_generator.py --proto-path=mcu_api/ --options-path=mcu_api -f mcu_api/mcu_api.options mcu_api.proto
mcu_api.proto:259:5: "pb_byte_t" is not defined.
mcu_api.proto:260:5: "pb_byte_t" is not defined.
mcu_api.proto:261:5: "pb_byte_t" is not defined.
```

Snippet from proto file is here:

```
message this_message_t
{
    pb_byte_t field_num = 1;
    pb_byte_t this_message_set1 = 2;
    pb_byte_t this_message_set2 = 3;
```

PetteriAimonen, 2024-02-07:
The .proto files must conform to the Google Protobuf format, which does not include custom types such as `pb_byte_t`.

You can use `int32` or `uint32` and override the generated C type size with option `[(nanopb).int_size = IS_8]`
See this example:

https://github.com/nanopb/nanopb/tree/master/tests/intsizes

shersey-locus, 2024-02-07:
OK, so the documentation talking about pb_byte_t and some other types in pb.h doesn't mean what it seemed to mean; those aren't actually supported field types. It might be worth adding a note to that effect, as the listing of types in pb.h immediately follows the description of things one can do in the options file. Makes its very easy to read it as allowable types.

closed 2024-02-07 by shersey-locus

PetteriAimonen, 2024-02-07:
It's part of the C API, not the protobuf side. Maybe the documentation is confusing in some ways, but I don't know how I could make it more clear.

shersey-locus, 2024-02-07:
Probably just me, then...