nanopb/nanopb #943: platformio generator fails due to bad protobuf package name

jeffc, 2024-02-28:
`platformio_generator.py` fails to build because it tries to `import protobuf` on [line 18](https://github.com/nanopb/nanopb/blob/master/generator/platformio_generator.py#L18) instead of `import google.protobuf`. I'm not totally sure when the package got renamed, but the rest of the .py files in the `generator/` directory use the new `google.` name.

I'm on archlinux using python 3.11.5 if that matters

PetteriAimonen, 2024-02-29:
Looks like it has just been wrong from the start, and would always execute the `pip` command (which would then say "requirement already satisfied".

PetteriAimonen, 2024-02-29:
Should be fixed now, you can use this in `platformio.ini` until new release is published:
  
    lib_deps =
        Nanopb=https://github.com/nanopb/nanopb/

closed 2024-09-19 by PetteriAimonen