nanopb/nanopb #945: How can the "oneof_callback" be utilized?

zhou-xiao-kun, 2024-03-01:
The predicament I am currently facing bears a striking resemblance to the demo(tests/oneof_callback/decode_oneof.c).
`typedef struct _SPM_HumanMachineInterface {
pb_size_t which_hmi_msg_type;
union {
bool reserved;
SPM_Touch_Report touch_report;
}hmi_msg_type;
}SPM_HumanMachineInterface;`
`typedef struct _SPM_Touch_Report {
    pb_callback_t event;
} SPM_Touch_Report;`
Although I have composed the global function SPM_Touch_Report_callback, it appears that its execution is not taking place as intended.
[image]
[image]
During the decoding process, I have inserted two log statements. The first log statement is successfully displayed, while the second one fails to appear. Based on this observation, I infer that the corresponding callback function might not have been executed. Could I have overlooked something that could be causing this issue?

PetteriAimonen, 2024-03-01:
To enable the name binding of `SPM_Touch_Report_callback`, you'll need an option in your .proto file (or in .options).
Either or combination of these will work:

* `[(nanopb).callback_datatype = "int"]` where the datatype can be anything you want, it is available in the message struct and at field->pData to store any information from the callback. When custom type is given, the callback function is automatically assigned to `Messagename_callback`.
* `[(nanopb).callback_function = "SPM_Touch_Report_callback"]` This will keep the default `pb_callback_t` datatype, but assign custom function name to be called instead of `pb_default_field_callback`.

When using name bound callbacks, you can omit the `oneof_callback` entirely. It's only necessary if you want to set function pointers in old-style `pb_callback_t` structure.

After this, you should see the declaration of the function name in generated `.pb.h` file.

zhou-xiao-kun, 2024-03-01:
Thank you for your prompt response. I will give it a try and close this question. thanks so much.

closed 2024-03-01 by zhou-xiao-kun