nanopb/nanopb #948: 指针未定义行为，空指针或者无效指针，导致Crash

hoodblock, 2024-03-14:
**源代码**

> `static bool safe_read_bool(const void *pSize)
{
    const char *p = (const char *)pSize;
    size_t i;
    for (i = 0; i < sizeof(bool); i++)
    {
        if (p[i] != 0)
            return true;
    }
    return false;
}`

_函数内部并没有对传入的指针进行有效性检查，可能会导致未定义行为。如果传入了空指针或者无效指针，代码可能会出现错误。

**是否合理解决**

`static bool safe_read_bool(const void *pSize)
{
    if (pSize == NULL) {
        return false; // 指针为空，返回false
    }
    const unsigned char *p = (const unsigned char *)pSize;
    size_t i;
    for (i = 0; i < sizeof(bool); i++)
    {
        if (p[i] != 0)
            return true;
    }
    return false;
}`

PetteriAimonen, 2024-03-14:
I don't understand whatever language this is.

hoodblock, 2024-03-18:
[image]
iOS Xcode develop use Pod
It should be caused by a null pointer