nanopb/nanopb #949: Crash on startup

hylke35, 2024-03-14:
I'm getting a strange crash for my iOS app on startup, this crash doesn't always occur, only sometimes.

This bit of code seems to be the issue, (specifically p[i]):
```
static bool safe_read_bool(const void *pSize)
{
    const char *p = (const char *)pSize;
    size_t i;
    for (i = 0; i < sizeof(bool); i++)
    {
        if (p[i] != 0)
            return true;
    }
    return false;
}
```

It's giving me the following error `Thread 1: EXC_BAD_ACCESS (code=1, address=0x1f01ee8ec)`

Based on the stacktrace the issue seems to be due to Firebase attempting to start something. What could be the reason for this, how can I rectify it? I've been busy for the past few days and can't seem to find a solution.

Stacktrace:
[image]

hylke35, 2024-03-14:
Just found [this](https://github.com/firebase/firebase-ios-sdk/issues/12457), seems to be a known issue at Firebase and they released an update for this issue. I'll update my packages, hopefully that resolves the issue.

hylke35, 2024-03-14:
Issue is resolved, see above comment.

closed 2024-03-14 by hylke35

hylke35, 2024-03-14:
> > Issue is resolved, see above comment.
> 
> I would wait for 10.23.0 release, as they fixed something in [10.22.1 that affects Xcode 15.3](https://firebase.google.com/support/release-notes/ios#version_10221_-_march_7_2024), which is not public on [CocoaPods](https://cocoapods.org/) (the latest public version being 10.20.0) and I would update the sdk to 10.20.0 and use Xcode 15.2 until then instead

Thanks for the tip, I'll check it out!

pinguluk, 2024-03-14:
Sorry, I meant 10.22.0, not 10.20.0

hoodblock, 2024-03-18:
> 在使用 Xcode 15.2 之前

> Issue is resolved, see above comment.

This kind of problem occurs occasionally, too headache, or just appeared in the last half month

Fever905, 2024-03-22:
Updated today to 10.23.0 and is still happening.  Would be a huge ordeal to have to downgrade to lower Xcode version!!

pinguluk, 2024-03-22:
> Updated today to 10.23.0 and is still happening. Would be a huge ordeal to have to downgrade to lower Xcode version!!

Have you verified the pods versions from platforms/ios? Because it's possible that if you just update the versions from package.json, it won't automatically update the pods that are already added.

Fever905, 2024-03-22:
Yes I ran pod update and it updated them.  Well weird, i just ran it using xCode and not using command line and it worked.

pinguluk, 2024-03-22:
> Yes I ran pod update and it updated them. Well weird, i just ran it using xCode and not using command line and it worked.

What error do you receive from command line?

Fever905, 2024-03-25:
No error on the command line, it would just launch the app then immediately quit.  Because it was a command line run I can't see the console output.  Is there a way to see that?

pinguluk, 2024-03-25:
> No error on the command line, it would just launch the app then immediately quit. Because it was a command line run I can't see the console output. Is there a way to see that? I published my app to the appstore Sunday. The only issues I was having was with devices running iOS versions below 15.8 ... like I had a customer iPad running iOS 15.0 that can't launch the app. My ipad running iOS 15.8 was fine. Anything below that doesn't seem to be launching.

Add **/usr/lib/swift** as explained here https://help.syncfusion.com/flutter/pdf-viewer/how-to/resolve-library-not-loaded-rpath-libswiftcore-dylib-error
or create a empty .swift file in the project's root and it should fix the startup crash

pinguluk, 2024-03-25:
> No error on the command line, it would just launch the app then immediately quit. Because it was a command line run I can't see the console output. Is there a way to see that? I published my app to [...]

And yeah, you can run the app via Xcode in Release mode (as it only occurs in this mode) and see the crash logs in there

Fever905, 2024-03-25:
OK thanks.  I used this site called Lambdatest and i tested back to iPhone 6 plus running iOS 13 and it works fine.  So I am good.  I think I will keep building with XCode seems like I get more control that way.   My app is for Industrial oil sample taking.  I have around 2000 users and need to make sure it's running at all times!

pinguluk, 2024-03-25:
> OK thanks. I used this site called Lambdatest and i tested back to iPhone 6 plus running iOS 13 and it works fine. So I am good. I think I will keep building with XCode seems like I get more control [...]

Feel you, I also recommend Browserstack, I use it to test out an app (with 100k+ users) on multiple devices as well

Fever905, 2024-03-27:
Thank you for the help!  :)

Meng991003, 2024-08-21:
I am hitting the same issue when using with firebasex version 16.5.0, have tried all above solution but its still not working.

Environment info:  
cordova-ios: 7.0.1,
cordova-plugin-firebasex: 16.5.0,
pods: 1.15.2,
IOS_FIREBASE_SDK_VERSION: 10.24.0

eastsss, 2024-10-07:
issue still exists, firebase 10.29.0, if built by xcode 15.4..
no issue if built by xcode 15.0