nanopb/nanopb #951: Make CMake `nanopb` library target optional

pdgendt, 2024-03-21:
The background:
Since PR #939 the `nanopb` target is always created, if `find_package(Nanopb)` is called.
With #946 I allowed the zephyr RTOS to use this repository directly, but the RTOS has it's own toolchain and build system, which supports many compilers and architectures.

With both PRs merged, the RTOS failed to compile the `nanopb` sample for some toolchain configurations (see https://github.com/zephyrproject-rtos/zephyr/issues/70516).

I think, to remain somewhat backwards compatible, the CMake `nanopb` target should be made optional.

Any suggestions/remarks?
CC @zachdeibert

PetteriAimonen, 2024-03-21:
I agree that it should be fixed in some way.

It would be nice to add a github workflow to nanopb repo that builds a zephyr sample, so we can catch these faster.

zachdeibert, 2024-03-21:
It looks like the root of the problem is that CMake isn't detecting C11 support properly.  I checked `arm-zephy-eabi-gcc` on Windows from Zephr SDK 0.16.5-1 and it compiles fine with `-std=c11` and C11 language features used.  I also printed `CMAKE_C_COMPILE_FEATURES` and it shows CMake is only detecting C90 and C99 support, but not C11 (CMake 3.28.1).  Maybe as a workaround we should check if `c_std_11` is in `CMAKE_C_COMPILE_FEATURES` and if it is then drop the C11 flag (and probably define `PB_C99_STATIC_ASSERT`)?

pdgendt, 2024-03-21:
The discussion has been done before actually: https://github.com/zephyrproject-rtos/zephyr/issues/66670

[`Nanopb` integration](https://github.com/zephyrproject-rtos/zephyr/blob/main/modules/nanopb/CMakeLists.txt) in zephyr already creates a (zephyr) library for the common sources, so the need for a `nanopb` library target just isn't there.

So my suggestion would still be to have it opt-in or opt-out, whichever is preferred.

zachdeibert, 2024-03-22:
It should already be optional in that if you `find_package(Nanopb)` and never reference the generated targets, it should never attempt to build those targets.  However, it does create them and your use case shows that the CMake scripts are lacking C99 support, which makes creating the targets fail.  If we add C99 support to the CMake script, would that fix your issue?

pdgendt, 2024-03-22:
> If we add C99 support to the CMake script, would that fix your issue?

Yes, I think that would make it nicely backwards compatible.

closed 2024-04-26 by pdgendt