nanopb/nanopb #952: Allow fixed size enums in C

jaskij, 2024-03-21:
C23 has added fixed size enums, similar to C++. ([compiler exploerer](https://godbolt.org/z/zPM4EYhWc))

Sorry for posting this in an issue, but I'm at work and don't have the time for a full PR, so want to write it down before I forget.

The [guard `ifdef`](https://github.com/nanopb/nanopb/blob/671672b4d7994a9b07a307ae654885c7202ae886/generator/nanopb_generator.py#L442) should probably be changed to something like below. It needs two changes:

- check for C++11 or newer (C++98 doesn't support fixed size enums)
- allow those for C23 or newer

```c
#if (defined(__cplusplus) && (__cplusplus > 201103L)) || (defined(__STDC_VERSION__) && (__STDC_VERSION__ > 202311L))
```

PetteriAimonen, 2024-07-19:
Closely related to #961.

jaskij, 2024-07-19:
Damn, I wasn't thinking when I wrote the thing about the guard.

I think this can be closed in favor of #961

closed 2024-07-19 by jaskij