nanopb/nanopb #953: Protobuf 26.0 removes RegisterExtension

wak-google, 2024-03-23:
https://github.com/protocolbuffers/protobuf/commit/4ebba684c791416a9131dd01d4c5175a2beb3566

```
nanopb/generator/nanopb_generator.py -q -s packed_struct:0 -H.n.h -S.n.c metricblob.proto
Failed to import generator/proto/nanopb_pb2.py:
Traceback (most recent call last):
  File "nanopb/generator/proto/__init__.py", line 102, in load_nanopb_pb2
    from . import nanopb_pb2 as nanopb_pb2_mod
  File "nanopb/generator/proto/nanopb_pb2.py", line 22, in <module>
    google_dot_protobuf_dot_descriptor__pb2.FileOptions.RegisterExtension(nanopb_fileopt)
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: type object 'FileOptions' has no attribute 'RegisterExtension'

Traceback (most recent call last):
  File "nanopb/generator/nanopb_generator.py", line 66, in <module>
    nanopb_pb2 = proto.load_nanopb_pb2()
                 ^^^^^^^^^^^^^^^^^^^^^^^
  File "nanopb/generator/proto/__init__.py", line 128, in load_nanopb_pb2
    return nanopb_pb2_mod
           ^^^^^^^^^^^^^^
UnboundLocalError: cannot access local variable 'nanopb_pb2_mod' where it is not associated with a value
```

PetteriAimonen, 2024-03-23:
Can you post `nanopb_generator.py --version --verbose` to check if the protoc version matches the python-protobuf version in use?

wak-google, 2024-03-24:
That doesn't actually work because of the error above...

PetteriAimonen, 2024-03-24:
Heh, makes sense.

What about `python generator/proto/_utils.py`? That should print the versions too.

wak-google, 2024-03-24:
````
liblwp-protocol-https-perl/noble,now 6.13-1 all [installed,automatic]
libprotobuf-dev/noble,now 3.21.12-8ubuntu5 amd64 [installed]
libprotobuf-lite32/noble,now 3.21.12-8ubuntu5 amd64 [installed,automatic]
libprotobuf32/noble,now 3.21.12-8ubuntu5 amd64 [installed,automatic]
libprotoc32/noble,now 3.21.12-8ubuntu5 amd64 [installed,automatic]
libx11-protocol-perl/noble,now 0.56-9 all [installed,automatic]
protobuf-compiler/noble,now 3.21.12-8ubuntu5 amd64 [installed]
rpcsvc-proto/noble,now 1.4.2-0ubuntu6 amd64 [installed,automatic]

pip3: protobuf                  5.26.0
```

It shouldn't be super relevant that the system libraries are older though. The issue is they deleted this function in the 5.26 release.

PetteriAimonen, 2024-03-24:
The function call is generated by the protoc python generator.

Google seems to break protobuf-python compatibility quite frequently, so if you have mismatching version of protobuf-compiler vs. python-protobuf, it leads to issues like this.

pdgendt, 2024-05-03:
Also bumped into this issue https://github.com/zephyrproject-rtos/zephyr/pull/72266

PetteriAimonen, 2024-05-03:
@pdgendt Could you expand on the issue? How are you installing, what does `python generator/proto/_utils.py` print?

pdgendt, 2024-05-03:
> @pdgendt Could you expand on the issue? How are you installing, what does `python generator/proto/_utils.py` print?
> 
> Edit: Noticed some more info in [zephyrproject-rtos/zephyr#72265](https://github.com/zephyrproject-rtos/zephyr/issues/72265) Does `pip install --upgrade grpcio-tools protobuf` help?

I don't think it helps, I upgraded all packages using
```shell
pip install --upgrade -r zephyr/scripts/requirements.txt
```

Which also contains `grpcio-tools>=1.47.0`.

pdgendt, 2024-05-03:
> what does `python generator/proto/_utils.py` print?

```shell
$ python modules/lib/nanopb/generator/proto/_utils.py
Using grpcio-tools protoc from /home/pdgendt/dev/zephyr-main/.venv/lib/python3.11/site-packages/grpc_tools/protoc.py
/home/pdgendt/dev/zephyr-main/modules/lib/nanopb/generator/proto/_utils.py:34: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
libprotoc 26.1
protoc builtin include path: ['/home/pdgendt/dev/zephyr-main/modules/lib/nanopb/generator/proto', '/home/pdgendt/dev/zephyr-main/.venv/lib/python3.11/site-packages/grpc_tools/_proto']
Python version 3.11.7 (main, Jan 25 2024, 09:38:07) [GCC 11.4.0]
Using python-protobuf from /home/pdgendt/dev/zephyr-main/.venv/lib/python3.11/site-packages/google/protobuf/__init__.py
Python-protobuf version: 5.26.1
```

PetteriAimonen, 2024-05-03:
@pdgendt Thanks. Could you further post `nanopb_pb2.py`?

For some reason I'm not getting the same error even with the same versions of packages:

<pre>
petteri@raichu:~/nanopb$ venv/bin/python generator/proto/_utils.py
Using grpcio-tools protoc from /home/petteri/gitrepos/nanopb/venv/lib/python3.10/site-packages/grpc_tools/protoc.py
libprotoc 26.1
protoc builtin include path: ['/home/petteri/gitrepos/nanopb/generator/proto', '/home/petteri/gitrepos/nanopb/venv/lib/python3.10/site-packages/grpc_tools/_proto']
Python version 3.10.12 (main, Nov 20 2023, 15:14:05) [GCC 11.4.0]
Using python-protobuf from /home/petteri/gitrepos/nanopb/venv/lib/python3.10/site-packages/google/protobuf/__init__.py
Python-protobuf version: 5.26.1

petteri@raichu:~/nanopb$ venv/bin/python generator/nanopb_generator.py examples/simple/simple.proto 
Writing to examples/simple/simple.pb.h and examples/simple/simple.pb.c
</pre>

pdgendt, 2024-05-03:
> @pdgendt Thanks. Could you further post `nanopb_pb2.py`?

```python
# -*- coding: utf-8 -*-
# Generated by the protocol buffer compiler.  DO NOT EDIT!
# source: nanopb.proto
# Protobuf Python Version: 5.26.1
"""Generated protocol buffer code."""
from google.protobuf import descriptor as _descriptor
from google.protobuf import descriptor_pool as _descriptor_pool
from google.protobuf import symbol_database as _symbol_database
from google.protobuf.internal import builder as _builder
# @@protoc_insertion_point(imports)

_sym_db = _symbol_database.Default()

from google.protobuf import descriptor_pb2 as google_dot_protobuf_dot_descriptor__pb2

DESCRIPTOR = _descriptor_pool.Default().AddSerializedFile(b'\n\x0cnanopb.proto\x1a google/protobuf/descriptor.proto\"\xb9\x07\n\rNanoPBOptions\x12\x10\n\x08max_size\x18\x01 \x01(\x05\x12\x12\n\nmax_length\x18\x0e \x01(\x05\x12\x11\n\tmax_count\x18\x02 \x01(\x05\x12&\n\x08int_size\x18\x07 \x01(\x0e\x32\x08.IntSize:\nIS_DEFAULT\x12$\n\x04type\x18\x03 \x01(\x0e\x32\n.FieldType:\nFT_DEFAULT\x12\x18\n\nlong_names\x18\x04 \x01(\x08:\x04true\x12\x1c\n\rpacked_struct\x18\x05 \x01(\x08:\x05\x66\x61lse\x12\x1a\n\x0bpacked_enum\x18\n \x01(\x08:\x05\x66\x61lse\x12\x1b\n\x0cskip_message\x18\x06 \x01(\x08:\x05\x66\x61lse\x12\x18\n\tno_unions\x18\x08 \x01(\x08:\x05\x66\x61lse\x12\r\n\x05msgid\x18\t \x01(\r\x12\x1e\n\x0f\x61nonymous_oneof\x18\x0b \x01(\x08:\x05\x66\x61lse\x12\x15\n\x06proto3\x18\x0c \x01(\x08:\x05\x66\x61lse\x12#\n\x14proto3_singular_msgs\x18\x15 \x01(\x08:\x05\x66\x61lse\x12\x1d\n\x0e\x65num_to_string\x18\r \x01(\x08:\x05\x66\x61lse\x12\x1b\n\x0c\x66ixed_length\x18\x0f \x01(\x08:\x05\x66\x61lse\x12\x1a\n\x0b\x66ixed_count\x18\x10 \x01(\x08:\x05\x66\x61lse\x12\x1e\n\x0fsubmsg_callback\x18\x16 \x01(\x08:\x05\x66\x61lse\x12/\n\x0cmangle_names\x18\x11 \x01(\x0e\x32\x11.TypenameMangling:\x06M_NONE\x12(\n\x11\x63\x61llback_datatype\x18\x12 \x01(\t:\rpb_callback_t\x12\x34\n\x11\x63\x61llback_function\x18\x13 \x01(\t:\x19pb_default_field_callback\x12\x30\n\x0e\x64\x65scriptorsize\x18\x14 \x01(\x0e\x32\x0f.DescriptorSize:\x07\x44S_AUTO\x12\x1a\n\x0b\x64\x65\x66\x61ult_has\x18\x17 \x01(\x08:\x05\x66\x61lse\x12\x0f\n\x07include\x18\x18 \x03(\t\x12\x0f\n\x07\x65xclude\x18\x1a \x03(\t\x12\x0f\n\x07package\x18\x19 \x01(\t\x12\x41\n\rtype_override\x18\x1b \x01(\x0e\x32*.google.protobuf.FieldDescriptorProto.Type\x12\x19\n\x0bsort_by_tag\x18\x1c \x01(\x08:\x04true\x12.\n\rfallback_type\x18\x1d \x01(\x0e\x32\n.FieldType:\x0b\x46T_CALLBACK\x12\x13\n\x0binitializer\x18\x1e \x01(\t*i\n\tFieldType\x12\x0e\n\nFT_DEFAULT\x10\x00\x12\x0f\n\x0b\x46T_CALLBACK\x10\x01\x12\x0e\n\nFT_POINTER\x10\x04\x12\r\n\tFT_STATIC\x10\x02\x12\r\n\tFT_IGNORE\x10\x03\x12\r\n\tFT_INLINE\x10\x05*D\n\x07IntSize\x12\x0e\n\nIS_DEFAULT\x10\x00\x12\x08\n\x04IS_8\x10\x08\x12\t\n\x05IS_16\x10\x10\x12\t\n\x05IS_32\x10 \x12\t\n\x05IS_64\x10@*Z\n\x10TypenameMangling\x12\n\n\x06M_NONE\x10\x00\x12\x13\n\x0fM_STRIP_PACKAGE\x10\x01\x12\r\n\tM_FLATTEN\x10\x02\x12\x16\n\x12M_PACKAGE_INITIALS\x10\x03*E\n\x0e\x44\x65scriptorSize\x12\x0b\n\x07\x44S_AUTO\x10\x00\x12\x08\n\x04\x44S_1\x10\x01\x12\x08\n\x04\x44S_2\x10\x02\x12\x08\n\x04\x44S_4\x10\x04\x12\x08\n\x04\x44S_8\x10\x08:E\n\x0enanopb_fileopt\x12\x1c.google.protobuf.FileOptions\x18\xf2\x07 \x01(\x0b\x32\x0e.NanoPBOptions:G\n\rnanopb_msgopt\x12\x1f.google.protobuf.MessageOptions\x18\xf2\x07 \x01(\x0b\x32\x0e.NanoPBOptions:E\n\x0enanopb_enumopt\x12\x1c.google.protobuf.EnumOptions\x18\xf2\x07 \x01(\x0b\x32\x0e.NanoPBOptions:>\n\x06nanopb\x12\x1d.google.protobuf.FieldOptions\x18\xf2\x07 \x01(\x0b\x32\x0e.NanoPBOptionsB\x1a\n\x18\x66i.kapsi.koti.jpa.nanopb')

_globals = globals()
_builder.BuildMessageAndEnumDescriptors(DESCRIPTOR, _globals)
_builder.BuildTopDescriptorsAndMessages(DESCRIPTOR, 'nanopb_pb2', _globals)
if not _descriptor._USE_C_DESCRIPTORS:
  _globals['DESCRIPTOR']._loaded_options = None
  _globals['DESCRIPTOR']._serialized_options = b'\n\030fi.kapsi.koti.jpa.nanopb'
  _globals['_FIELDTYPE']._serialized_start=1006
  _globals['_FIELDTYPE']._serialized_end=1111
  _globals['_INTSIZE']._serialized_start=1113
  _globals['_INTSIZE']._serialized_end=1181
  _globals['_TYPENAMEMANGLING']._serialized_start=1183
  _globals['_TYPENAMEMANGLING']._serialized_end=1273
  _globals['_DESCRIPTORSIZE']._serialized_start=1275
  _globals['_DESCRIPTORSIZE']._serialized_end=1344
  _globals['_NANOPBOPTIONS']._serialized_start=51
  _globals['_NANOPBOPTIONS']._serialized_end=1004
# @@protoc_insertion_point(module_scope)
```

PetteriAimonen, 2024-05-03:
Ok, that exactly matches what I have. It also does not have `RegisterExtension` call, like would be generated by e.g. `libprotoc 3.12.4`:

    nanopb_fileopt.message_type = _NANOPBOPTIONS
    google_dot_protobuf_dot_descriptor__pb2.FileOptions.RegisterExtension(nanopb_fileopt)

Full file: [nanopb_pb2.py_protoc_3.12.4.txt](https://github.com/nanopb/nanopb/files/15197537/nanopb_pb2.py_protoc_3.12.4.txt)

I strongly suspect that for some reason there is a different version of protoc involved when you get the error.
Do you get the same error if you try `python generator/nanopb_generator.py examples/simple/simple.proto` from command line?

pdgendt, 2024-05-03:
> I strongly suspect that for some reason there is a different version of protoc involved when you get the error. Do you get the same error if you try `python generator/nanopb_generator.py examples/simple/simple.proto` from command line?

Hmn, no:

```shell
$ python modules/lib/nanopb/generator/nanopb_generator.py zephyr/samples/modules/nanopb/src/simple.proto
Writing to zephyr/samples/modules/nanopb/src/simple.pb.h and zephyr/samples/modules/nanopb/src/simple.pb.c
```

pdgendt, 2024-05-03:
From the log I see that the system `/usr/bin/protoc` is used, how can I force to use `grpc_tools/protoc.py` instead?

PetteriAimonen, 2024-05-03:
grpc_tools protoc is the primary choice if `import grpc_tools.protoc` succeeds.
I think the sequence of events here is:

1. Zephyr build rules call `/usr/bin/protoc`, which in turn calls `protoc-gen-nanopb`. That the system protoc is used at this point is largely irrelevant.
2. Nanopb generator attempts to import `nanopb_pb2.py`, but fails to due to version mismatch.
3. Nanopb generator attempts to rebuild `nanopb_pb2.py`. It is not clear whether this fails completely or if it ends up using wrong version of `protoc` for some reason. If the latter, it would probably be due to Python search path not including the grpcio installation for some reason.

Barabas5532, 2024-05-04:
I encountered the same issue in my CI pipeline running the generator using the provided cmake function while having protoc version 3.12.4 and grpcio-1.63.0 grpcio-tools-1.63.0 protobuf-5.26.1 installed.

On my development machine running the same python packages and protoc 25.3, I didn't have any issues. I use nanopb as a git submodule so the same nanopb version is used in both environments.

I managed to work around by using an updated `requirements.txt` that looks like this:

```
protobuf~=4.25
grpcio-tools
```

This installs "grpcio-1.63.0 grpcio-tools-1.62.2 protobuf-4.25.3" which works in my CI setup.

It looks like protobuf tracks breaking changes in the first version number. I think the same change (possibly using `~=4.0`) should be made to the bundled requirements.txt until the code can be updated to work with the breaking changes in version 5.16 of the protobuf python package.

PetteriAimonen, 2024-05-04:
@Barabas5532 Problem is that I'm not aware of any code that could be updated to work with the new version. All of the code with incompatibilities is generated by protoc.

I hate how protobuf-python keeps breaking its generated file API, leading to this kind of problems when systems have multiple protoc versions. But I'm not sure whether switching to pure-protobuf or python-betterproto would actually make things much better.

I currently don't get any problems with a clean venv and latest requirements.txt:

<pre>
petteri@raichu:~/nanopb$ python3 -m venv venv

petteri@raichu:~/nanopb$ venv/bin/pip install -r requirements.txt 
Collecting protobuf
  Using cached protobuf-5.26.1-cp37-abi3-manylinux2014_x86_64.whl (302 kB)
Collecting grpcio-tools
  Using cached grpcio_tools-1.63.0-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (2.3 MB)
Requirement already satisfied: setuptools in ./venv/lib/python3.10/site-packages (from grpcio-tools->-r requirements.txt (line 2)) (59.6.0)
Collecting grpcio>=1.63.0
  Using cached grpcio-1.63.0-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (5.6 MB)
Installing collected packages: protobuf, grpcio, grpcio-tools
Successfully installed grpcio-1.63.0 grpcio-tools-1.63.0 protobuf-5.26.1

petteri@raichu:~/nanopb$ venv/bin/python generator/nanopb_generator.py examples/simple/simple.proto 
Writing to examples/simple/simple.pb.h and examples/simple/simple.pb.c
</pre>

pdgendt, 2024-05-06:
> grpc_tools protoc is the primary choice if `import grpc_tools.protoc` succeeds. I think the sequence of events here is:
>
> 1. Zephyr build rules call `/usr/bin/protoc`, which in turn calls [...]

For 1. it's the `find_program` call in `extra/FindNanopb.cmake` that defaults to the system `/usr/bin/protoc`, this generates the `nanopb_pb2.py` that is incompatible with later calls.

PR #958 (2024-05-06): Prefer `generator/protoc` in CMake
When building with CMake we should prefer to use the `protoc` executable that uses grpcio-tools.

Fixes #953
closed 2024-05-06 by PetteriAimonen