nanopb/nanopb #960: Fixed length/ count isn't enforced in python (both Serialize & Deserialize)

idan-ben-ami, 2024-05-08:
Given the following .proto
```proto
syntax = "proto2";

import "nanopb.proto";  // Import the nanopb extension

message TestMessage {
    required bytes t1 = 1 [(nanopb).max_size = 8, (nanopb).fixed_length = true];
    repeated int32 t2 = 2 [(nanopb).max_count = 4, (nanopb).fixed_count = true];
}
```

I manage to instantiate and serialize & deserialize the following:
```py
from test_pb2 import TestMessage
# t1 is less than 8 bytes
demo = TestMessage(t1=b"Hello", t2=[1,2,3])
raw = demo.SerializeToString()
other = TestMessage()
other.ParseFromString()
```

It also works on larger inputs than `max_size`/`max_count`.

Though, in C/C++, the deserialization of `raw` throws the required exception.

The generation of test_pb2 is straight forward: `protoc --python_out=. test.proto`

PetteriAimonen, 2024-05-08:
Yes, nanopb-specific options do not affect other protobuf libraries.

idan-ben-ami, 2024-05-08:
Right, I forgot that nanopb is strictly C/C++.

Thanks

closed 2024-05-08 by idan-ben-ami