nanopb/nanopb #961: Incorrect generated data_size when using (nanopb_enumopt).int_size

mrjoboto, 2024-05-13:
Hi there,
I’ve been experimenting with using the `(nanopb_enumopt).int_size` option to control the size of the generated enum. However, I’ve been running into issues during encode/decode when I set the enum size to 1 byte. Specifically, it seems like the message descriptor that gets generated for the enum field has an incorrect data_size value that doesn’t respect the int_size option used. But the generated struct does respect the int_size option. This leads to the encode function accessing memory beyond the end of the message structure. I tried to put together a contained example that shows the problem:
```
syntax = "proto3";

import "nanopb.proto";
option (nanopb_fileopt).long_names = false;

enum Foo {
  option (nanopb_enumopt).int_size = IS_8;
  kBar = 0;
  kBaz = 1;
}

message EnumContainer {
  Foo foo = 1;
}
```
The generated structures are what I would expect:
```
/* Enum definitions */
typedef enum _Foo
#ifdef __cplusplus
 : uint8_t
#endif
{
    kBar = 0,
    kBaz = 1
} Foo;

/* Struct definitions */
typedef struct _EnumContainer {
    Foo foo;
} EnumContainer;
```

However, the generated message descriptor seems to result in an inaccurate data_size for the enum field:
```
EnumContainer message = EnumContainer_init_zero;
message.foo = Foo::kBaz;

pb_field_iter_t iter;
pb_field_iter_begin_const(&iter, EnumContainer_fields, (void*)&message);
printf("%lu, %hu\n", sizeof(message.foo), iter.data_size);
```

For me this prints “1,4” - the size of the field is 1 byte, but the data_size value is 4 bytes, and data_size is used during encoding to iterate over a void* pointing into the message structure, leading it to access unrelated memory.

Any advice you have to offer would be most appreciated! I'm using nanopb 0.4.8 and clang 15.0.0 targetting arm64-apple-darwin22.6.0.

PetteriAimonen, 2024-05-13:
Hmm, are you perhaps generating the protocol definitions into a `.pb.c` file, but compiling your user code as `.cc` file?
Can you try putting definitions to `.pb.cc` instead?

I think `#ifdef __cplusplus` should be removed from the generator. Now C23 supports `: uint8_t` in C mode, and in any case it is better to have compiler error than runtime error.

mrjoboto, 2024-05-13:
Thanks for the prompt reply!

> Hmm, are you perhaps generating the protocol definitions into a `.pb.c` file, but compiling your user code as `.cc` file? Can you try putting definitions to `.pb.cc` instead?

Yes, this is the case - I am generating the protocol definitions into a .pb.c file and compiling the user code as cpp. I think I see what you're saying - the field info is getting generated in a context where `__cplusplus` is not defined and the enum size is 4 bytes.
Manually changing the definition files to .cc resolves the issue. Is there a nanopb option to output .cc files (or otherwise handle this situation)? I couldn't find one.

PetteriAimonen, 2024-05-13:
Yes, `--source-extension=.cc` for nanopb_generator.py, or `--nanopb_opt="--source-extension=.cc"` when called through protoc.

oleid, 2024-07-02:
I think I got a similar/related issue.

I'm using a protobuf file similar to the following:

```.proto
syntax = "proto3";

import "nanopb.proto";

option (nanopb_fileopt).max_count = 10;
option (nanopb_fileopt).int_size = IS_16; // generate 16 bit integers by default
option (nanopb_fileopt).proto3_singular_msgs = true;  // Prevents has_* fields generated in structs

enum DeviceMode {
  INITIAL = 0;
  STANDBY = 1;
  OPERATIONAL = 2;
  WARNING = 3;
  FAULT = 4;
  SELFUPDATE = 6;
  SELFTEST = 7;
}
```

The generated code looks as follows:

```.h
/* Automatically generated nanopb header */
/* Generated by nanopb-0.4.8 */

#ifndef PB_UARTCOMM_PB_H_INCLUDED
#define PB_UARTCOMM_PB_H_INCLUDED
#include <pb.h>

#if PB_PROTO_HEADER_VERSION != 40
#error Regenerate this file with the current version of nanopb generator.
#endif

/* Enum definitions */
typedef enum _DeviceMode
#ifdef __cplusplus
 : uint16_t
#endif
{
    DeviceMode_INITIAL = 0,
    DeviceMode_STANDBY = 1,
    DeviceMode_OPERATIONAL = 2,
    DeviceMode_WARNING = 3,
    DeviceMode_FAULT = 4,
    DeviceMode_SELFUPDATE = 6,
    DeviceMode_SELFTEST = 7
} DeviceMode;
```

As you can see, for C++ the integer width of the enum is set to 16 bit.
I have a little helper library implemented in C that reads the data to an instance of the struct, which is created in C++ code.
The C++ code is some desktop test code, which talks to the micro controller via UART.
So  the problem originates due to ABI incompatibility between C and C++ when using these headers.
I know how to work-around the issue. I just left a note here, as I think this is a serious footgun when using custom integer widths.

jaskij, 2024-07-19:
Looking at the code, putting the size of the enum behind `#ifdef __cplusplus` is just plain wrong, as it would result in different ABI depending on whether the code is parsed as C or C++.

The only sane option I can see here is to remove the `#ifdef` and document that this only works if:

- you are using C23
- everything is compiled as C++, 11 or higher
- the compiler used has an extension allowing declaring enum sizes in older versions of C

When it comes to changes in the generator itself, the change would be to just not output that `#ifdef __cplusplus` and let the compiler error.

Maybe someone else has a better idea?

jaskij, 2024-07-19:
There is another option, although I do not know enough about nanopb's internals to say how viable it would be.

Do not use `int_size` for enums, and introduce a new option specifically for them.

PetteriAimonen, 2024-07-20:
Summarizing the things brought up:

* Support for this in C is fairly new in compilers.
* It's better to fail compilation than to emit code where ABI is different between C and C++

As such I think best course of action is to:

* Separate the enum size option to `enum_intsize`. This avoids accidentally setting it with wildcards.
* No `#ifdef` guards, if compiler doesn't support it, compilation will fail.
* Separate the enum_intsize test case so that it is run only with C++ compiler for now.
* `docs/migration.md` needs updates

jaskij, 2024-07-20:
I believe this should be tested, to make sure there is no ABI breakage.

#993 changes the CI to add tests on Ubuntu 24.04, which comes with GCC 14. That should enable tests.

I propose we add two tests, one using C++11, the other using C23, and gate the C23 test behind compiler support in SCons.

closed 2024-09-19 by PetteriAimonen