nanopb/nanopb #962: Required fields for proto3

canton7, 2024-05-15:
Hi,

We've been using nanopb extensively on a fairly large embedded project, for Ethernet/BLE communications with a number of devices, for file storage, and for logging. It's performed really well, and we've benefited a lot from it. Thank you!

There's one thing (with two aspects) which has been a bit of a source of pain however, and I'd like to suggest a way to improve it.

Proto3 (as you know) removed support for `required`, meaning that all fields which have a `message` type are implicitly optional, and so `nanopb` generates a corresponding `has_xxx field`.

However, on our embedded system many of these fields are not considered optional: they need to be present, and the message should be rejected if they are not. This means that we need to write validation. However, this validation code is best written at the point that the message is deserialized (it's otherwise hard to pass the correct errors back up through the call chain from where the message is actually used, C being what it is, and quite often messages are stored and then actually used later on). That's fine, but it means that the code which checks the presence of a `has_xxx` field is far removed from the code which actually accesses that field, and so it's hard to keep the two in sync.

The other aspect is the reverse: the embedded code will always populate some fields, however it needs to remember to set the `has_xxx` field at every point. This adds significant noise, and is also easy to forget (which has led to a few bugs). We also end up with some confusion over whether a field is actually semantically required to be set by the embedded side or not.

My suggestion is to add a custom option which works the same as `required` for proto2: fields with this option set will have no corresponding `has_xxx` field, will always be serialized, and nanopb will return an error if they are not present on deserialization.

I'm aware of the reasons that `required` fields were removed in proto3, and I generally agree with them. However, I don't think they apply here, as a "required" option will only apply to nanopb and not to other parts of the system:

* For deserialization, it's functionally the same as manually writing code which validates that the corresponding `has_xxx` field is `true`.
* For serialization, it only states that the embedded device expects to send the field. Other devices in the ecosystem will be unaware of the option, and will therefore treat the field as optional. The option can therefore be removed later without breaking those devices.

We could achieve this by moving back to proto2, however the proto3 changes around enums and packed arrays are desirable. And, using proto2's `required` of course has downsides.

From my reading of the nanopb code, adding this should be technically easy, as it just re-uses the support for `required` in proto2. I'm happy to have a shot at implementing this if you give the go-ahead.

What to you think?

PetteriAimonen, 2024-05-15:
I wonder how this would work for fields with zero value? Presence would be indistinguishable from zero value.

Proto3 scalar fields should not have generated `has_` except for submessages. It is possible to disable `has_` for submessages also by setting `proto3_singular_msgs`, in which case a message is considered present if any of its field are non-zero.

I'm not strictly against this feature, but I wonder whether it is a bit niche. Considering it would be generator-only change, the impact is very small, so I'm ok with including it as long as it has a test case.

Possibly it could be made as a generic `rules_override` (the term used by nanopb) or `label_override` (the term used in `descriptor.proto`). It's a bit of "you need to know what you are doing" feature, but similar to already existing `type_override`.

canton7, 2024-05-15:
> I wonder how this would work for fields with zero value? Presence would be indistinguishable from zero value.
>
> Proto3 scalar fields should not have generated has_ except for submessages. It is [...]

Sorry, I should have been clearer. I'm only thinking about submessages -- scalar fields don't have this issue, as you say. I think that also removes the question around fields with zero value?

> It is possible to disable has_ for submessages also by setting proto3_singular_msgs, in which case a message is considered present if any of its field are non-zero.

That's certainly useful. However it doesn't cover the case of rejecting the message on reception/deserialization if (submessage) fields which the embedded side considers required are not present.

The serialization side is also subtly different: there are submessage fields which both sides currently consider required, and the non-embedded (receiving) side will raise an error if the field is not present in messages sent from the embedded side. So omitting the field is semantically different from including it, even if all of its fields are zero. Forcing the non-embedded side treat a missing submessage field as all-zeros means it's harder to evolve the protocol in future to actually remove that field.

> Possibly it could be made as a generic rules_override (the term used by nanopb) or label_override (the term used in descriptor.proto). It's a bit of "you need to know what you are doing" feature, but similar to already existing type_override.

This is slightly over my head, but if I'm interpreting you correctly, that would be something like `(nanopb).label_override = FieldDescriptorProto.LABEL_REQUIRED`? Sure, I don't see why not. That allows `LABEL_OPTIONAL` which could conceivably have some use, and `LABEL_REPEATED` which I suppose wouldn't cause an issue (as it could still interact with any `max_size` etc)?

PetteriAimonen, 2024-05-15:
Regarding last point, yeah. I think the syntax would be `(nanopb).label_override = LABEL_OPTIONAL`, as the enum type would be resolved through the definition in `nanopb.proto`. Test case could be just added to `tests/options`.

PR #963 (2024-05-15): Add label_override field option
This lets proto3 files define submessage fields which nanopb treats as required (even though proto3 doesn't support 'required'). This means that nanopb won't generate a has_xxx field, will reject messages where this field is missing, and will always serialize the field. However, other consumers of the proto file won't see the field as required, and are therefore safe to it being removed in future.

To do this, use:

    Submessage some_submessage = 1 [(nanopb).label_override = LABEL_REQUIRED];

This mechanism also lets users change whether nanopb treats the fiel [... cut]
commit 9e37f612 (2024-05-15): Add label_override field option

This lets proto3 files define submessage fields which nanopb treats as
required (even though proto3 doesn't support 'required'). This means that
nanopb won't generate a has_xxx field, will reject messages where this
field is missing, and will always serialize the field. However, other
consumers of the proto file won't see the field as required, and are
therefore safe to it being removed in future.

To do this, use:

    Submessage some_submessage = 1 [(nanopb).label_override = LABEL_REQUIRED];

This mechanism also lets users change whether nanopb treats the fie [... cut]
closed 2024-05-15 by PetteriAimonen