nanopb/nanopb #964: Allow struct alignment specification, or make alignment of packed structs consistent across compilers

mrjoboto, 2024-05-15:
This is more of a feature request, not a bug report. I was looking at macros related to struct packing in pb.h, and saw that for some compilers it uses the `pragma(pack(push, 1))` syntax (which specifies both packing and alignment), but for gcc/clang, it uses `__attribute__((packed))` syntax (which specifies only packing, not alignment, AFAIK).

It would be useful for my application if there was a nanopb generator option to specify struct alignment. Or, alternatively, GCC and clang support the `pragma(pack(push, 1))` syntax, so it would be possible to use that syntax consistently across all known compilers (but that might cause some unexpected issues for existing GCC/clang codebases).

PetteriAimonen, 2024-05-15:
Might make sense to have a generic option that would allow including any kind of attribute in the struct spec.

mrjoboto, 2024-05-15:
That would be very handy!

canton7, 2024-05-16:
`__attribute__((packed))` implicitly uses an alignment of 1.

If there's no packing, then the size of the struct could be any integer, therefore an array of such structs may contain a struct at any alignment, therefore the struct has to have an alignment of 1.

Have a play with https://ideone.com/fqBag8.

mrjoboto, 2024-05-20:
Yeah, I realize I misspoke when I implied we'd expect inconsistent alignment behavior here. Although I can't find any guarantees in the GCC documentation that `__attribute((packed))` always implies an alignment of 1 in all situations (it just says each member is "placed to minimize the memory required").

Part of the reason for the confusion was that we are seeing inconsistent behavior with how GCC/Clang handle `__attribute__(packed)` vs `#pragma pack` (some [bug](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=90452) [reports](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=93910) if you're curious). A relevant compiler warning doesn't get generated when using `#pragma pack`.

Anyway, I think this supports the idea that a generic attribute option would be nice, to give users maximum control.