nanopb/nanopb #967: Pure ESP-IDF framework nanopb support?

anthonyhoegberg, 2024-05-28:
Hello, looking around i see that there is PlatformIO support for this library out there right now, however how would one get it to work with just using plain esp-idf using the components dir or something similar, im quite new to this stuff and nanopb has its own project definition so i cant just paste it in and expect it to work ofc

Any support/help would be appreciated

PetteriAimonen, 2024-05-28:
How about just copy the source code files and add them to build? I'm not familiar with esp-idf, but that's the way it should work with any build system without anything special.

playduck, 2024-05-31:
I've implemented nanopb into an esp-idf project. It's pretty trivial, here's a brief tutorial:
I'm currently using the very newest ESP-IDF v5.4 - if you're on something different then YMMV

1. Create a new idf component (e.g. create a directory at `/components/your-protobuf-component/` (the `components` directory is at the same level as your `main` and `build` folders)
2. Generate your `.pb.c` and `.pb.h` files using a call to `nanopb/generator/nanopb_generator.py`
   2.1 Move the generated files to this component directory
   2.2 Move the nanopb files to this directory. You NEED `pb.h`, `pb_common.h`/`pb_common.c`. As well as either one or both `pb_encode.h`/`pb_encode.c` and/or `pb_deencode.h`/`pb_encode.c`.
3. Create a new `CMakeLists.txt:
```CMake
file(GLOB SOURCE_FILES "*.c" "*.h")

idf_component_register(SRCS ${SOURCE_FILES}
                    INCLUDE_DIRS "."
)
```

> Note: Globbing the source files is much easier than manually specifying them, however there are [some caveats to be aware of!](https://stackoverflow.com/questions/1027247/is-it-better-to-specify-source-files-with-glob-or-each-file-individually-in-cmak) You could also manually specify all the `.c` and `.h` files here.

4. That's it! You can now `#include "your-generated-protobuf.pb.h"`. I'd recommend writing a wrapper module for your specifics in the component as well as that will make it more portable, but how you use it is entirely up to you.

Note: I've included nanopb as a git submodule of my project and automated the entirely compilation and the moving of files to the component with a script. That makes it much easier and cleaner to re-generate and update. You may also be able to simlink the non-generated files like `pb.h` to the component directory - haven't tried that though.

anthonyhoegberg, 2024-06-05:
Thank you for your response, it seems like manually putting the files is the way to go here, although automatically including it somehow would feel nicer.  I will close this issue now. Thanks.

closed 2024-06-05 by anthonyhoegberg