nanopb/nanopb #970: Byte Array output generation

Bgorkhe, 2024-06-04:
Hello,

I am using this struct as example but this feature might also be available to get the raw byte array from the structs.

```
typedef struct test_s {
uint8_t sync;
    uint8_t size;
    uint8_t data[4];
    uint32_t crc;
} pb_packed test_t;

```
If I use nano protobuf and generate the bytes it comes as 
`0x8 0x5 0x10 0xc 0x1a 0x4 0x20 0xc 0x2d 0x5a 0x20 0x93 0xeb 0x94 0xd1 0x5 `

Whereas if I just use this struct with uint8_t casting pointer to generate byte array and get the data back its generated as 
`0x05 0x0c 0x20 0x0c 0x2d 0x5a 0x93 0x35 0x25 0x5a`

Probably I am missing how these bytes are generated but is there is a way to get raw bytes without any extra bytes which will be feature request if its not available.

Thank you,
BGorkhe

PetteriAimonen, 2024-06-04:
I don't understand the question. The entire point of nanopb as a library is to encode the data into bytes in the protobuf format.

As you say, if you want raw bytes, you can use `memcpy()` on any struct.

PetteriAimonen, 2024-06-04:
https://protobuf.dev/programming-guides/encoding/

Bgorkhe, 2024-06-04:
Thank you @PetteriAimonen for writing nanopb. I will close I think I got my misunderstanding clear now.
- BGorkhe.

closed 2024-06-04 by Bgorkhe