nanopb/nanopb #972: CI build failing for Ubuntu 20.04

PetteriAimonen, 2024-06-09:
For some reason the binary package build is failing with this error:

<pre>
[2912] Error loading Python lib '/home/runner/work/nanopb/nanopb/nanopb-0.4.8-32-g4474bd3-linux-x86/generator-bin/_internal/libpython3.8.so.1.0': dlopen: /home/runner/work/nanopb/nanopb/nanopb-0.4.8-32-g4474bd3-linux-x86/generator-bin/_internal/libpython3.8.so.1.0: cannot open shared object file: No such file or directory
</pre>

* [Failing build](https://github.com/nanopb/nanopb/actions/runs/9281881983/job/25538736455)
* [Working build](https://github.com/nanopb/nanopb/actions/runs/9281881973/job/25538792392)

But I haven't figured out what has changed and how to fix it.

jaskij, 2024-07-11:
Looking at the [`python3` package in Ubuntu 20.04](https://packages.ubuntu.com/focal/python3), and it's [changelog](https://changelogs.ubuntu.com/changelogs/pool/main/p/python3-defaults/python3-defaults_3.8.2-0ubuntu2/changelog), they bumped the version from 3.8.0 to 3.8.2 in late February.

I don't know if GH's runners actually keep in line with Ubuntu's packages, but looking at the error message, it's probably related, even if the timing of this seems weird.

PetteriAimonen, 2024-07-12:
@jaskij Thanks for looking into it.

In the "Setup image" section both runs report docker image version "ubuntu-20.04 20240526.1.0"
Diffing the log files I also couldn't find any version differences, just that pip seems to be installing some packages in different order.

A fix would be great even if we don't figure out what changed. Typically pyinstaller just needs some command line switches to fix problems like this. I think the workflows will run in a forked repo when you enable them in the repo settings. I tried some things in `dev_ci_fix` branch but to no avail. Will probably need to figure out how to get reverse ssh into the github runner - I seem to recall that it should be possible.

jaskij, 2024-07-12:
Looking at the logs manually, there is one more difference between the two jobs you linked in the issue. The working one has this line in job setup:

```
Uses: nanopb/nanopb/.github/workflows/binary_packages.yml@refs/heads/master (4474bd35bd39de067f0532a1b19ce3aed9ed9807)
```

Seems like some non-deterministic edge case is biting you here.

----

As a diagnostic, I'd add `python --version` and `pip --version` somewhere in your CI script to check the installed versions of your tools.

----

That said, from what I know, you should not be using `pip` anyway - it's interactions with system libraries are known to be an issue. Even with `--user`. This build *will* break on newer Ubuntu versions (I believe somewhere around 23). Why I mention it here is that using a venv may help with this issue even without knowing the root cause.

----

For my own convenience I have downloaded the logs, stripped them of timestamps, and diffed. There seem to be more differences worth looking into. I'm attaching all three files.

[fail.txt](https://github.com/user-attachments/files/16195504/fail.txt)
[success.txt](https://github.com/user-attachments/files/16195506/success.txt)
[log-diff.txt](https://github.com/user-attachments/files/16195511/log-diff.txt)

jaskij, 2024-07-12:
Looking more at the logs, I can see that PyInstaller uses different Python shared libraries between the two versions too.

Failed:

```
3216 INFO: Using Python shared library: /lib/x86_64-linux-gnu/libpython3.8.so
```

Succeeding:

```
2908 INFO: Using Python shared library: /lib/x86_64-linux-gnu/libpython3.8.so.1.0
```

And then it errors out when checking protoc version of all things.

jaskij, 2024-07-18:
Doing some experimentation, I managed to confirm this is an upstream bugs. I don't know what exactly happened (my bet is on either GHA or PyInstaller). Doing some testing, I managed to get a [working job](https://github.com/jaskij/nanopb/actions/runs/9990278684/job/27610649552). I then did some more test commits, reverted them, and a job with *the exact same repository contents* [failed](https://github.com/jaskij/nanopb/actions/runs/9990687522/job/27611897088).

PetteriAimonen, 2024-07-18:
@jaskij Thanks a lot! Seems to work at least initially, I'll trigger a full weekly flow and see how it goes.

PR #990 (2024-07-18): ci: binary_packages: add actions/setup-python
This *should* fix #972 , although I am not completely sure.

As I mentioned in https://github.com/nanopb/nanopb/issues/972#issuecomment-2236295006 this seems to be a really weird issue. My guess is that this is due to GitHub not correctly sanitizing the systems between jobs.

The problems I mentioned in #989 caused me problems with testing, but by running some jobs in a different branch, it seems that adding the `actions/setup-python@v5` action should do the trick.

I have had three jobs in a row succeed:

- https://github.com/jaskij/nanopb/actions [... cut]
closed 2024-07-18 by PetteriAimonen