nanopb/nanopb #973: Usage of STRIP causes error as if ${NANOPB_PLUGIN_OPTIONS} is empty

philam2001, 2024-06-10:
Noticed an issue with a codebase that uses nanopb that I'm working on with the new update because of this STRIP command. I researched it a little bit, it seems like if the ${NANOPB_PLUGIN_OPTIONS} that is part of the arguments in the STRIP command is empty, this can cause issues for the command to only see one argument. A thing that could be added is a check to see if the option is empty or not, another thing is to put quotation marks on the variable. If this is deemed an legitimate issue I could also put up a PR myself, I don't really know how contributions to open source codebases work.

PetteriAimonen, 2024-06-11:
@gsurkov Any thoughts?

gsurkov, 2024-06-11:
I think that surrounding the variable in quotation marks is the way to go because not only it fixes the problem, it can also prevent other potential bugs. And I'm pretty sure the STRIP command does the check for empty string internally.

Changes are proposed in #974, sorry for the inconvenience my changes have caused.

philam2001, 2024-06-11:
No worries, looks good!

closed 2024-06-11 by philam2001