nanopb/nanopb #976: Custom include prefix?

jaskij, 2024-06-14:
For use cases where there are multiple protocol files referencing each other. There will be situations when `foo.pb.h` needs to include `bar.pb.h`. The thing is, I want neither of the files directly in the include search path. I want them in `protocol/`, so instead of `#include "bar.pb.h"`, the line should be `#include "protocol/bar.pb.h"`.

This can, of course, be manually edited, but manually editing generated files is never a good idea.

If there is such an option, I'm sorry, I was unable to find it in the documentation.

PetteriAimonen, 2024-06-15:
This should work in this way, which is standard across protobuf platforms:

1. In your `.proto` file, use `import "protocol/bar.proto";` (including the directory name)
2. Run protoc or nanopb_generator from the upper folder using command such as `nanopb_generator protocol/bar.proto`.

Then the generated files will also have the directory name in their include directories.

If, for some reason, you want a different folder structure for nanopb than for other protobuf libraries, you can use `'--generated-include-format=#include "foo/%s"'`

jaskij, 2024-06-18:
Thank you for the explanations, works as intended.

closed 2024-06-18 by jaskij