nanopb/nanopb #981: Code style request: Add typehints to python3 code, add docstrings to C and Python Code

danielappiagyei-bc, 2024-07-05:
**What should the feature do?**
Hi, I was wondering if you've ever considered [doxygen style comments](https://www.doxygen.nl/manual/docblocks.html) to your C function declarations? Preferably this style would be added to all function _declarations_:
```
/**
 * @brief <description>
 * @param stream <description>
 * @param buf <description>
 * @param count <description>
 *
 * @return <description>
 */
static bool checkreturn buf_read(pb_istream_t *stream, pb_byte_t *buf, size_t count)
```

We use them on my firmware team and although a lot of your excellent code is self-documenting (thank you for creating this project!), I'm finding it would be handy (as I read through nanopb source code) to reference the docstring for any important info or expectations. It's neat too because any modern IDE (VSCode, CLion, Sublime, etc.) neatly gives a preview of the docstring (documentation-string) when you hover over the name  (see below screenshot): 
[image]

The same goes for using docstrings in python3 (with the exception that the common docstring style is something like [this](https://sphinx-rtd-tutorial.readthedocs.io/en/latest/docstrings.html), as opposed to doxygen-style). I see you use a docstring variant in the generator but you don't have the `param <param>:` fields nor [typehints](https://docs.python.org/3.5/library/typing.html). 
If these sound attractive to you then I could definitely volunteer to make some PRs to add them? The only issue is the `typing` module in python was only adding in python 3.5 and I don't know what minimum python version you support.

Thanks again for creating an awesome tool!
**In what situation would the feature be useful?**

See above!

PetteriAimonen, 2024-07-06:
Currently I'm still keeping Python 2 support, though that will probably be dropped in next major release (0.5 or 1.0, whatever the name will be, probably in 2025).

Regarding doxygen comments, I would want some community feedback on this. Personally I'm not a fan of them, because in my opinion they are much longer and more annoying to read.

danielappiagyei-bc, 2024-07-06:
Thanks for the quick reply! 
Regarding python, I would only care for the typehints in the `python3`-only gen script `generator/nanopb_generator.py`, as opposed to `generator/nanopb_generator.py2`. Does that change your opinion on that? As long as you have dictated that python3.5 or higher is required to use it then there will be no issues.  Also are you open to adding docstrings to python code?

> Regarding doxygen comments, I would want some community feedback on this. Personally I'm not a fan of them, because in my opinion they are much longer and more annoying to read.

That's totally fair, I think that's a common complaint of them. Looking at your code again, I'm seeing you do have a lot of functions commented, just not in a doxygen-style manner, that's great! Let's use the generic term of "doctsrings" to refer to any kind of function-description comments. What would help me understand the code faster would be the following:
1. requiring **every function**, no matter how big or small, to have docstrings. Looking at `pb_decode.c`, only some of the file-static functions have docstrings
2. Requiring the meaning of the return type to be documented more thoroughly than just "true on success; false otherwise". E.g., in what cases does `pb_read()` return true and in what cases does it return false? It looks like it returns true if:
`count` is 0 (i.e., you're not requesting it to read any bytes) or some other case I can't immediately tell from the `return true` at the last line of the function. There's a myriad of cases where it returns false.
3. Requiring any important invariants to  be documented. E.g., looking at `buf_read()`, the `stream` param is not allowed to be NULL but `buf` is. Also the function always returns true, which is interesting. 

The above three can be done using any (including your) docstring format, not just doxygen. Additionally, since a lot of the args for functions would have the same docstring (e.g., 10+ functions in `pb_decode.c` have the same `pb_istream_t *stream` argument), we can just omit them from any docstring with the option of documenting them in one place in the file (not necessarily as a certain function's docstring). 

Thank you again for your consideration and for creating this tool! You've led this project for over a decade so you definitely know best; these suggestions are just minor things that I'm seeing would help me understand and master the code a whole lot faster as someone who's just now looking at our current version to upgrade it to the latest version. :pray:

PetteriAimonen, 2024-07-10:
Thanks for the feedback, especially regarding what you would want to see in the comments. I'll keep that in mind even though I'm not going to be working on it immediately.

If you are interested in working on this, pull requests are welcome. Adding comments for every function using current style would be a good idea.

jaskij, 2024-07-11:
A good argument for Doxygen, or a different similar tool, is single source of truth. Poking around a little, you currently generate the API reference with Pandoc from a manually written `reference.md`.

With Doxygen, you could generate DocBook or manpage output, then run it through the existing Pandoc pipeline.

On another note, you can put Doxygen comments in both `.c` files and `.h` files - it doesn't really care. Depending on where you find them less annoying.

Personally, I never look at those comments, didn't even realize they were there, I always go to your online API reference. I don't even remember the last time I looked at Nanopb's source code.