nanopb/nanopb #987: Cut nanopb release containing bzlmod changes

tpudlik, 2024-07-16:
Would it be possible to cut a release (I guess it would be 0.4.9?) containing https://github.com/nanopb/nanopb/pull/950? This would allow adding nanopb to the [Bazel Central Registry](https://github.com/bazelbuild/bazel-central-registry), and so make it very straightforward to depend on nanopb from Bazel projects using bzlmod.

PetteriAimonen, 2024-07-17:
I agree that it would be time for a new release, but it will require resolving at least #972 first and preferrably also #961, #956. It may be a while before I have time and motivation to dive into it.

tpudlik, 2024-07-17:
Thank you for listing the blockers! I'll take a look at the issues and see if I can help out with them.

jaskij, 2024-07-19:
@PetteriAimonen can you pin this issue to highlight the blockers for the next release?

PetteriAimonen, 2024-07-19:
Pinned & added milestone https://github.com/nanopb/nanopb/milestone/3

I went through the issue list and seems those are all that seem important. #887 I didn't consider high priority before, but with later Python deprecations it has become so.

PetteriAimonen, 2024-09-19:
0.4.9 released

closed 2024-09-19 by PetteriAimonen