nanopb/nanopb #997: Emit deprecation warnings

jaskij, 2024-07-29:
Protobuf allows marking a field as deprecated

```proto
int32 old_field = 6 [deprecated = true];
```

With C23 introducing [standard attributes](https://en.cppreference.com/w/c/language/attributes) it would be nice to emit that attribute on struct members marked as such in the `.proto`.

I am unsure whether it would be better to have an option specifically for `[[deprecated]]`, or an option enabling general C23 support.

PetteriAimonen, 2024-07-30:
C23 support itself could be detected in preprocessor. It could be something like:

    PB_DEPRECATED(int foobar);

and

    #ifndef PB_DEPRECATED
    #  ifdef __whatever_c23_defines
    #    define PB_DEPRECATED(x) [[deprecated]] x
    #  else
    #    define PB_DEPRECATED(x) x
    #  endif
    #endif

which would then allow things like compiler-specific deprecation support.

Though in many cases it might make sense to have a way to easily set all deprecated fields to `FT_IGNORE` in generator.

jaskij, 2024-07-30:
Hmm... that's an option, but personally I'd have it be prefix, not function, style, like this:

```c
#ifndef PB_DEPRECATED
    #if __STDC_VERSION__ > 202311L
        #define PB_DEPRECATED [[deprecated]]
    #elif defined(__GNUC__)
        #define PB_DEPRECATED __attribute__((deprecated))
    #else 
        #define PB_DEPRECATED
    #endif
#endif
```

> Though in many cases it might make sense to have a way to easily set all deprecated fields to FT_IGNORE in generator.

That's a great idea too.

P.S.

You can enable syntax highlighting by add the name of the language without space after the opening backtics of the code block.

PetteriAimonen, 2024-09-03:
Hmm, I ran into a bit of a gotcha in trying to implement this: the default `PB_BIND` refers to every field of the message, so it will cause the deprecation error/warning regardless if user code actually uses it.

I think it would require compiler-specific pragmas to selectively disable the warning around the PB_BIND area.

PetteriAimonen, 2024-09-03:
Implemented the simpler approach of a `discard_deprecated` option.

closed 2024-09-19 by PetteriAimonen