nanopb/nanopb #1000: ld: library 'nanopb' not found

Muhaymin-Dvago-Developer, 2024-08-08:
I've run into this problem after integration of webengage Rich Notification. I've tried everything again and again but unable to fix it.

pod deintegrate
pod cache clean
rm -rf podfile.lock
rm -rf Pods
rm -rf build 
pod install && pod update && pod install --repo-update

I've done everything i could but didn't fixed this issue. Below is the screenshots and the error logs shown in the xcode (15.1) 

**Error**
Ld /Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator/myProject.app/myProject normal (in target 'myProject' from project 'myProject')
    cd /Users/Developer/Desktop/myProjectApp-1/ios
    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++ -Xlinker -reproducible -target x86_64-apple-ios13.4-simulator -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/SDKs/iPhoneSimulator17.2.sdk -Os -L/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Intermediates.noindex/EagerLinkingTBDs/Release-iphonesimulator -L/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator -L/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/SDKs/iPhoneSimulator17.2.sdk/usr/lib/swift -L/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/swift/iphonesimulator -L/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator/Google-Maps-iOS-Utils -L/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator/react-native-google-maps -L/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator/react-native-maps -L/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/swift/iphonesimulator -L/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/SDKs/iPhoneSimulator17.2.sdk/usr/lib/swift -L/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/SDKs/iPhoneSimulator17.2.sdk/usr/lib/swift -L/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator/Google-Maps-iOS-Utils -L/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator/react-native-google-maps -L/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator/react-native-maps -L/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/swift/iphonesimulator -L/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/SDKs/iPhoneSimulator17.2.sdk/usr/lib/swift -L/Users/Developer/Desktop/myProjectApp-1/ios/Pods -L/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator/nanopb -F/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Intermediates.noindex/EagerLinkingTBDs/Release-iphonesimulator -F/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator -F/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator/Base64 -F/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator/CodePush -F/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator/DoubleConversion -F/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator/FirebaseABTesting -F/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator/FirebaseCore -F/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator/FirebaseCoreExtension -F/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator/FirebaseCoreInternal -F/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator/FirebaseCrashlytics -F/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator/FirebaseDynamicLinks -F/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator/FirebaseInstallations -F/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator/FirebaseMessaging -F/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator/FirebasePerformance -F/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator/FirebaseRemoteConfig -F/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator/FirebaseRemoteConfigInterop -F/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator/FirebaseSessions -F/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator/FirebaseSharedSwift -F/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator/GoogleDataTransport -F/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator/GoogleUtilities -F/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator/JWT -F/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator/Mixpanel-swift -F/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator/MixpanelReactNative -F/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator/PromisesObjC -F/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator/PromisesSwift -F/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator/RCT-Folly -F/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator/RCTDeprecation -F/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator/RCTTypeSafety -F/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator/RNCAsyncStorage -F/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator/RNCClipboard -F/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator/RNDateTimePicker -F/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator/RNDeviceInfo -F/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator/RNFBAnalytics -F/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator/RNFBApp -F/Users/Developer/Library/Developer/Xcode/DerivedData/myProject-cjrycdzzwdpgiqgtvhzleeoxyvdq/Build/Products/Release-iphonesimulator/RNFBCrashlytics -F/Users/Developer/Library/Develop [... cut ...]
  installer.pods_project.build_configurations.each do |config|
    config.build_settings["EXCLUDED_ARCHS[sdk=iphonesimulator*]"] = "arm64"
  end

  installer.pods_project.targets.each do |target|
    if target.name == 'WebEngageAppEx'
      target.build_configurations.each do |config|
        config.build_settings['SWIFT_INSTALL_OBJC_HEADER'] = 'NO'
      end
    end
  end
end

**xcode Settings**
[image]
[image]
[image]
[image]

Please let me know if i can provide anything else for more clarification

Thanks.

PetteriAimonen, 2024-08-08:
I think you may want to contact the framework provider, I don't think this is relevant to the nanopb core library.

closed 2024-08-08 by PetteriAimonen

Muhaymin-Dvago-Developer, 2024-08-09:
> I think you may want to contact the framework provider, I don't think this is relevant to the nanopb core library.

@PetteriAimonen Can you help me in this i don't know what else to do i've done a lot of tries to fix this but unable to do it ?

PetteriAimonen, 2024-08-09:
I cannot help, I have no knowledge of Xcode or pods.