nanopb/nanopb #1001: Possible optimizations for messages with large "oneofs" and proto 3 definitions without defaults

mrtzcspr, 2024-08-13:
Hi,

we are using relatively big "oneof"-messages in our proto-files and I found that Nanopb spends quite some time in iterating over these "oneof" fields. I think this is mainly caused by the necessity of setting default values that may not be zero. But this somehow feels unnecessary because at maximum only one of these fields could be set at any time and it should be possible by looking at the `which_*`-struct member, to determine which field actually needs to be checked for default values. Also as we are only using "proto 3" which have no default values at all.

I modified the "simple"-example by adding 3 "oneof"s and respective 50 "oneof"s to the "SimpleMessage". But I do not initalize them, so that still the complete message can be represented as `message.lucky_number = 13` and will encode to just the 2 bytes `0x08, 0x0D`.

The modified example can be found here:  https://github.com/mrtzcspr/nanopb/commit/5ff5bc77c98d5add20dff445e663561833afafe4

When using `pb_decode_noinit` the "SimpleMessage" with 50 "oneof"s gets decoded ~20times faster than when using the normal `pb_decode`function. Also with `pb_decode_noinit` there seems to be no performance penalty compared to a "SimpleMessage" with only 3 "oneof"s.

Unfortunately there is currently no `pb_encode_noinit` or `pb_encode_nodefaults` variant, that could be used to tell Nanopb not to try to set the default values when encoding. In my benchmarking the "SimpleMessage" with 50 "oneof"s ist ~8 times slower than the 3 "oneof"s message. 

Could you consider adding such a function? Or maybe also (if possible) add some logic that looks at the `which_*`-struct member and only iterates over the necessary fields?

## My measurements:

### Encoding (100.000 loops)

[image]

### Decoding (100.000 loops)

[image]

PetteriAimonen, 2024-08-14:
I haven't tried to repeat your benchmarks, but here are my comments based on how the code works.

On decoding, default values are set only for the specific oneof that was received. The other messages occupy the same memory space in C union, so initializing them would be pointless. However, prior to decoding all fields are iterated through to check whether they are part of oneof or something else, so it makes sense that `pb_decode_noinit()` saves time.

> Unfortunately there is currently no pb_encode_noinit or pb_encode_nodefaults variant, that could be used to tell Nanopb not to try to set the default values when encoding.

I'm not sure what this means. The encoding does not modify the struct and shouldn't care about default values (apart from the proto3 zero-value check). The main encoding loop descends only into the oneof that is indicated by the `which_` field. However, it does iterate through all fields in the top-level message to check whether they should be encoded (e.g. if they are part of oneof or something else).

In the end I think this boils down to the overhead in field iteration functions in pb_common.h. They are a tradeoff between flash usage and execution speed.

----

Regarding your specific case, I would recommend doing the top level encoding and decoding separately using `pb_encode/decode_tag()` and `pb_encode/decode_submessage()`. This allows more flexiblity in how you store the structs in memory and avoids the need to go through descriptor array to find the specific message.

For example, you could have a function like this:

```
bool send_response(pb_ostream_t *stream, pb_size_t oneof_field_number, const pb_msgdesc_t *fields, const void *struct)
{
    return pb_encode_tag(stream, PB_WT_STRING, oneof_field_number)
      && pb_encode_submessage(stream, fields, struct);
}
```

and then call it like this:

     MySubMessage msg = { ... };
     send_response(stream, TopLevelMessage_mysubfield_tag, MySubMessage_fields, &msg);

This has exactly same encoded representation as the oneof.

mrtzcspr, 2024-08-14:
Hello Petteri,
thanks for your fast and extensive response.

I just benchmarked encoding a oneof with 3 members compared with a oneof with 50 possible members. With `pb_encode` the smaller oneof is approx. ~4times faster encoding the same message.

With your recommended approach from above the number of oneofs does not affect the result and compared to the normal encode its nearly 6 times faster for the proto-file with 50 oneof members:

[image]

So this looks like a good approach that I'll try to implement for my actual situation of double nested oneofs 😄 . 

Do I miss something or should it be relatively straight forward to do the submessage encode in one pass, by just encoding the submessage and later prepending the size? Is there maybe even public code that does this? Having gotten the benchmarking fever right now I see another 2x runtime reduction waiting there..

PetteriAimonen, 2024-08-14:
Yeah, you can do that on multiple levels.

One trick can be to start the stream a few bytes after the beginning of the buffer, that way you have space to add the size tag.

closed 2024-09-04 by PetteriAimonen