nanopb/nanopb #1004: Couldn't compile generated vaildate.proto from protoc-gen-validate

KindDragon, 2024-09-04:
validate.proto use reserved names in C/C++ `enum`, `float`, `double`...
https://github.com/bufbuild/protoc-gen-validate/blob/main/validate/validate.proto

```
arm_gnueabihf_gcc/gen/moira/proto/validate/validate.pb.h:29:16: warning: declaration does not declare anything
   29 |     float const;
      |                ^
arm_gnueabihf_gcc/gen/moira/proto/validate/validate.pb.h:60:17: warning: declaration does not declare anything
   60 |     double const;
      |                 ^
arm_gnueabihf_gcc/gen/moira/proto/validate/validate.pb.h:91:18: warning: declaration does not declare anything
   91 |     int32_t const;
      |                  ^
arm_gnueabihf_gcc/gen/moira/proto/validate/validate.pb.h:122:18: warning: declaration does not declare anything
  122 |     int64_t const;
      |                  ^
arm_gnueabihf_gcc/gen/moira/proto/validate/validate.pb.h:153:19: warning: declaration does not declare anything
  153 |     uint32_t const;
      |                   ^
arm_gnueabihf_gcc/gen/moira/proto/validate/validate.pb.h:184:19: warning: declaration does not declare anything
  184 |     uint64_t const;
      |                   ^
arm_gnueabihf_gcc/gen/moira/proto/validate/validate.pb.h:215:18: warning: declaration does not declare anything
  215 |     int32_t const;
      |                  ^
arm_gnueabihf_gcc/gen/moira/proto/validate/validate.pb.h:246:18: warning: declaration does not declare anything
  246 |     int64_t const;
      |                  ^
arm_gnueabihf_gcc/gen/moira/proto/validate/validate.pb.h:277:19: warning: declaration does not declare anything
  277 |     uint32_t const;
      |                   ^
arm_gnueabihf_gcc/gen/moira/proto/validate/validate.pb.h:308:19: warning: declaration does not declare anything
  308 |     uint64_t const;
      |                   ^
arm_gnueabihf_gcc/gen/moira/proto/validate/validate.pb.h:339:18: warning: declaration does not declare anything
  339 |     int32_t const;
      |                  ^
arm_gnueabihf_gcc/gen/moira/proto/validate/validate.pb.h:370:18: warning: declaration does not declare anything
  370 |     int64_t const;
      |                  ^
arm_gnueabihf_gcc/gen/moira/proto/validate/validate.pb.h:401:15: warning: declaration does not declare anything
  401 |     bool const;
      |               ^
arm_gnueabihf_gcc/gen/moira/proto/validate/validate.pb.h:407:24: warning: declaration does not declare anything
  407 |     pb_callback_t const;
      |                        ^
arm_gnueabihf_gcc/gen/moira/proto/validate/validate.pb.h:502:24: warning: declaration does not declare anything
  502 |     pb_callback_t const;
      |                        ^
arm_gnueabihf_gcc/gen/moira/proto/validate/validate.pb.h:551:18: warning: declaration does not declare anything
  551 |     int32_t const;
      |                  ^
arm_gnueabihf_gcc/gen/moira/proto/validate/validate.pb.h:598:35: warning: declaration does not declare anything
  598 |     google_protobuf_Duration const;
      |                                   ^
arm_gnueabihf_gcc/gen/moira/proto/validate/validate.pb.h:631:36: warning: declaration does not declare anything
  631 |     google_protobuf_Timestamp const;
      |                                    ^
arm_gnueabihf_gcc/gen/moira/proto/validate/validate.pb.h:669:29: error: two or more data types in declaration specifiers
  669 |         validate_FloatRules float;
      |                             ^~~~~
arm_gnueabihf_gcc/gen/moira/proto/validate/validate.pb.h:669:34: warning: declaration does not declare anything
  669 |         validate_FloatRules float;
      |                                  ^
arm_gnueabihf_gcc/gen/moira/proto/validate/validate.pb.h:670:30: error: two or more data types in declaration specifiers
  670 |         validate_DoubleRules double;
      |                              ^~~~~~
arm_gnueabihf_gcc/gen/moira/proto/validate/validate.pb.h:670:36: warning: declaration does not declare anything
  670 |         validate_DoubleRules double;
      |                                    ^
In file included from ../../../third-party/components/nanopb/Nanopb/pb.h:86,
                 from arm_gnueabihf_gcc/gen/moira/proto/sberhome/ODM_devices/v1/ble_pairing.pb.h:6:
arm_gnueabihf_gcc/gen/moira/proto/validate/validate.pb.h:681:28: error: two or more data types in declaration specifiers
  681 |         validate_BoolRules bool;
      |                            ^~~~
arm_gnueabihf_gcc/gen/moira/proto/validate/validate.pb.h:681:32: warning: declaration does not declare anything
  681 |         validate_BoolRules bool;
      |                                ^
arm_gnueabihf_gcc/gen/moira/proto/validate/validate.pb.h:685:32: error: expected ‘{’ before ‘;’ token
  685 |         validate_EnumRules enum;
      |                                ^
arm_gnueabihf_gcc/gen/moira/proto/validate/validate.pb.h:685:28: error: two or more data types in declaration specifiers
  685 |         validate_EnumRules enum;
      |                            ^~~~
arm_gnueabihf_gcc/gen/moira/proto/validate/validate.pb.h:686:9: error: unknown type name ‘validate_RepeatedRules’
  686 |         validate_RepeatedRules repeated;
      |         ^~~~~~~~~~~~~~~~~~~~~~
```

PetteriAimonen, 2024-09-04:
Can you check what Google's protobuf C++ library does for these symbols? It would make sense for nanopb to do similar symbol renaming.

KindDragon, 2024-09-04:
It just add underscore at the end of reserved word https://github.com/protocolbuffers/protobuf/blob/c841627af4949957255089466353292431ad45a6/src/google/protobuf/compiler/cpp/helpers.cc#L61-L158

https://github.com/protocolbuffers/protobuf/blob/c841627af4949957255089466353292431ad45a6/src/google/protobuf/compiler/cpp/helpers.cc#L439-L444

[validate.zip](https://github.com/user-attachments/files/16871300/validate.zip)

PetteriAimonen, 2024-09-04:
Thanks, I'll add the same. I'll make it configurable but default-on, just in case someone relies on unmangled names.

PetteriAimonen, 2024-09-04:
(By the way, if you don't actually need validate.proto in your nanopb-based runtime but it is referred to by the shared .proto file, you can add the header to exclude list.)

ChapterSevenSeeds, 2026-02-01:
This just came up for me too. I started poking around and playing with the generator to see what a possible fix could look like. But, I figured I would see if there is any headway being made on this before doing anything further 😛