nanopb/nanopb #1006: Protobuf 5.28 runtime warning

pdgendt, 2024-09-16:
Since https://github.com/protocolbuffers/protobuf/pull/17241 got merged, the following warning is printed when using `nanopb`:

```
google/protobuf/runtime_version.py:112: UserWarning: Protobuf gencode version 5.27.2 is older than the runtime version 5.28.1 at nanopb.proto. Please avoid checked-in Protobuf gencode that can be obsolete.
```

An upstream issue has been created here: https://github.com/protocolbuffers/protobuf/issues/18096

PetteriAimonen, 2024-09-16:
This sounds like mismatched `grpcio-tools` and `python-protobuf` versions.

mrtzcspr, 2024-10-02:
@PetteriAimonen Are you aware that this also affects the latest release bin package? 

Steps to reproduce:

```bash
curl -OL https://jpa.kapsi.fi/nanopb/download/nanopb-0.4.9-linux-x86.tar.gz
tar -xf nanopb-0.4.9-linux-x86.tar.gz
cd nanopb-0.4.9-linux-x86/examples/simple/
 ../../generator-bin/nanopb_generator simple.proto
```
produces:
```
google/protobuf/runtime_version.py:112: UserWarning: Protobuf gencode version 5.27.2 is older than the runtime version 5.28.2 at nanopb.proto. Please avoid checked-in Protobuf gencode that can be obsolete.
Writing to simple.pb.h and simple.pb.c
```

PetteriAimonen, 2024-10-02:
@mrtzcspr I was not, thanks for letting me know. I guess I should lock the versions used for generating binary packages, though then I'd forget to update them..