nanopb/nanopb #1007: Allow truncation of 'bytes' fields

yperess, 2024-09-18:
# The problem
I have a client/server firmware that I'm working with where one of the endpoints might be updated less frequently than the other. They communicate in different platforms on I2C, eSPI, or HECI buses (with I2C being the bottleneck for throughput). In the model, the server may fire a GPIO to notify the client that a new status is available and the client will initiate the read as the controller of the bus. Here's a simplified message that might be read:

```proto
enum Flag {
  DATA_READY = 0;
  LOG_BUFFER_WATERMARK = 1;
  LID_OPEN = 2;
  ...
}

message Status {
  bytes flags = 1 [(nanopb).max_size=2];
  uint32 response_length = 2;
}
```

In this case the `flags` field will be 2 bytes. The issue is when the proto changes to include new `Flag` entries and the `flags` field needs another byte. At this point the client will read the `Status` but from the client's perspective there are 2 bytes available for flags, while the server sent 3 (I have a second mechanism in place to packetise the message so we don't just leverage `Status_size`).

Currently, when we get to `pb_dec_bytes()` we'll fail to decode because we only have 2 bytes available. From a product perspective this should be allowed, the server sent us new information that we don't know what to do with. This is no different than if the message looked like:

```proto
message Status {
  byte flag1 = 1;
  byte flag2 = 2;
  byte flag3 = 3;
  uint32 response_length = 4;
}
```

When the server adds flag3, the client would know to ignore it. While splitting up the bitmap into separate fields would solve the problem, it adds traffic to an already slow bus which I'd like to avoid.

# The proposal
Add a new option `bool allow_overflow` which would only apply to `bytes` fields with `max_size` and would allow decoding to truncate the overflow bytes if present.

Please let me know if this seems reasonable and I'll get working on adding it + unit tests.

PetteriAimonen, 2024-09-18:
I think this may be a bit too specific need to be included in the main codebase.

In general I think it would make sense to have the flags as uint32 instead of bytes, but I don't know your application requirements.

For using bytes but not caring about overflow, using a custom callback could work:

     bytes flags = 1 [(nanopb).callback_datatype = 'struct flags_bytes_t { uint8_t size; uint8_t bytes[2]; }',
                      (nanopb).callback_function = 'Status_callback'];

and on C side (untested code):

     bool Status_callback(pb_istream_t *istream, pb_ostream_t *ostream, const pb_field_t *field)
     {
         if (istream && field->tag == Status_flags_tag)
         {
             uint32_t size;
             flags_bytes_t *dest = field->pData;
             if (!pb_decode_varint32(stream, &size)) return false;

             if (size > sizeof(dest->bytes))
             {
                   dest->size = sizeof(dest->bytes);
                   if (!pb_read(stream, dest->bytes, dest->size) return false;
                   if (!pb_read(stream, NULL, size - dest->size) return false;
             }
             else
             {
                   dest->size = size;
                   if (!pb_read(stream, dest->bytes, dest->size) return false;
             }
         }

        return true;
     }

yperess, 2024-09-18:
> I think this may be a bit too specific need to be included in the main codebase.
>
> In general I think it would make sense to have the flags as uint32 instead of bytes, but I don't know your [...]

Thank you for that, I'll try this approach

yperess, 2024-09-18:
I was able to get things to work, thank you!

closed 2024-09-18 by yperess