nanopb/nanopb #1014: Bad default initializer values for messages with enum types

mhjartstrom, 2024-09-26:
When yielding Enum definitions in ProtoFile.generate_header() using NamingStyleC, \_\_str()__ will call enum_entry() which will return self.underscore(name).upper().

Soon afterwards, the initializer values for message structs are also yielded, but Field.get_initializer() will not make any special treatment for enum default values and hence will set inner_init = str(self.default).

Now, if the enum was defined with mixed case names of the type and/or entries, the initializer values will be illegal since the generated enum definitions in the header file is upper case.

Something as simple as adding the following to [nanopb_generator.py:874](https://github.com/nanopb/nanopb/blob/473fbd6098929b45f2ac6c6424d186bb6dfb12a8/generator/nanopb_generator.py#L874) will do, but there might be other side effects in doing so (?):

`elif self.pbtype in ('ENUM', 'UENUM'):`
`  inner_init = str(self.default).upper()`

PetteriAimonen, 2024-09-26:
Can you include a proto file that reproduces this issue?

mhjartstrom, 2024-09-26:
[Test.zip](https://github.com/user-attachments/files/17150378/Test.zip)

The attached file will typedef a my_enum_t enum with two upper case value names and a msg_t message. The #define MSG_INIT_DEFAULT will have an undefined camel case value set.

Please disregard my quick and dirty suggestion in the first message, it solved my immediate problem but does not work on the attached example (it will not underscore the upper case values correctly), so it feels as if some conversion of the Field to an Enum is required for this to work, otherwise it will not reproduce the same naming convention.

PR #1020 (2024-10-17): Fix inner enum naming with a default value
Fixes #1014 

The first commit adds a test that fails because it generates an undeclared identifier `MyEnum1_ENTRY_Second` in `MAIN_MESSAGE_INIT_DEFAULT` instead of the expected `MY_ENUM1_ENTRY_SECOND`. The second commit fixes it by utilizing the configured naming style for enum values. 

## Simple Example

Given the following _test.proto_ file:

```protobuf
syntax = "proto2";

enum MyEnum {
    ValueOne = 1;
    ValueTwo = 2;
}

message Msg {
    required MyEnum enum = 1 [default = ValueOne];
}
```

Running `python3 nanopb_generator.py --c-style` wou [... cut]
closed 2024-10-17 by PetteriAimonen