nanopb/nanopb #1017: inconcistencies around end of stream reading

inolen, 2024-10-03:
Hey @PetteriAimonen, wanted to bring this up while I ran into this issue with #1008.

Right now, the main decode loop looks like -

```
while (stream->bytes_left) {
  if (!decode_tag(&eof)) {
    if (eof) break;
    else return false;
  }
}

... handle misc errors, e.g. required fields ...
```

There's a couple of issues I noticed here - 

1. This doesn't behave consistent when callbacks are used. Streams not using callbacks will break out of this when bytes_left is 0 before attempting to call decode_tag and setting the "end-of-stream" error, meanwhile with callbacks the callback will be called, which will in turn set bytes_left to 0 / return false and ultimately set the "io error" error in pb_readbyte before the loop breaks. This means things like required errors won't be set correctly when callbacks are used as there is already an error on the stream set.

2. Callbacks setting bytes_left to 0 feels weird, since pb_decode.c otherwise manages that. Perhaps adopting the read() behavior and returning an integer of bytes read / 0 (eof) / -1 (error) would make handling this state easier.

3. Using stream->bytes_left as the loop condition feels flawed with callbacks who set the stream's size to some non-exact size. It feels like decode_tag could just return false indicating either an eof or error, and decode_inner could return stream->errmsg != NULL.

PetteriAimonen, 2024-10-04:
I agree that this should probably be reviewed. I think the current design is somewhat a result of #278.

inolen, 2024-10-04:
@PetteriAimonen what are the restrictions around modifying the callback interface (e.g. to return +n / 0 / -1) instead of a bool?

PetteriAimonen, 2024-10-04:
It affects user code, so it should only be done after good consideration. On the other hand, 1.0.0 would be the correct time to do the change if it is worthwhile.

Using signed return value also introduces annoying edge case if the data size exceeds (or claims to exceed, such as in malicious bytes payload) 2 GB.

inolen, 2024-10-05:
AFAIK, a ssize_t can always hold a size_t, right? The read() system call returns an ssize_t and takes an size_t for count.

PetteriAimonen, 2024-10-06:
To me it feels like the issues might be solvable without such a large change. I strongly prefer API stability when possible.

But I have to admit I haven't really looked into this yet.

inolen, 2024-10-06:
Totally understood re: stable API.

The fundamental problem here is that for each read, there are 3 possible results -

1. success
2. eof
3. error

With the bool, we can return only one. If a correct bytes_left was set for every stream, the bool would be sufficient as it could be used to determine eof leaving the bool to determine success / error. This is normally the case for buffer streams, but not for callbacks (e.g. in basic_stream/decode_stream it sets SIZE_MAX for the bytes_left).