nanopb/nanopb #1018: nanopb.proto as package

atelszewski, 2024-10-09:
Hello,

in the project I work on, it's expected that `nanopb.proto` declares package to be `nanopb`.
Then, nanopb options in `*.proto` files are referenced like so: `(nanopb.nanopb).max_size = 25`.

For the moment, this is accomplished by adding `package nanopb;` declaration to `nanopb.proto`.
However, I would like to switch our project to use the upstream and unmodified nanopb sources.

So my question is, is it possible to make `nanopb.proto` be exposed as `package nanopb`,
without source code changes?

Thank you for you assistance! :-)

--
Best regards,
Andrzej Telszewski

PetteriAimonen, 2024-10-09:
Because of the cumbersome `(nanopb.nanopb)`, I'm not planning to change the package name declared in the proto.

You could consider using a separate `.options` file, in which case you do not need to reference `nanopb.proto`.

atelszewski, 2024-10-09:
Oh, I guess I didn't make myself clear enough.

I received `.proto` files that already import `nanopb.proto`.
Preferably, these `.proto` files should _not_ be _changed_.

But I guess then it's not possible to achieve my goal without modifying `nanopb.proto` to contain `package nanopb`?

My hope was that, that maybe nanopb generator can somehow _add_ this directive _on the fly_, but now I think that's not the case. Is that correct?

Thanks!

PetteriAimonen, 2024-10-10:
So what is your goal, for `nanopb.proto` to have `package nanopb` without having `package nanopb`? I admit I cannot understand.

The options are processed by Google Protobuf's `protoc` before they get to nanopb generator, it will be the one that complains if the name is not what it expects it to be.

closed 2024-10-10 by atelszewski