nanopb/nanopb #1019: CMake and how to apply options file to all proto files (i.e. globally)

atelszewski, 2024-10-09:
Hello,

I'm looking for a way to apply single options file to all `.proto` files.
I'm using CMake and `NANOPB_GENERATE_CPP()`.

I found that it's possible to apply options _per_ `.proto` file, if the options file name corresponds to the `.proto` file name.

But how can I have a single _options_ file be applied for all `.proto` files?

Thanks for your help! :-)

--
Best regards,
Andrzej Telszewski

PetteriAimonen, 2024-10-10:
I think something like this would work:

    list(APPEND NANOPB_OPTIONS "--options-file=myfile.options")

The full list of generator command line options can be had with `nanopb_generator.py --help`.

atelszewski, 2024-10-10:
That did the trick, thank you!

(Apparently I already tried this solution before asking here, but failed due to some hiccups.)

Thanks!

closed 2024-10-10 by atelszewski