nanopb/nanopb #1023: Support creating a typedef for oneof unions

jackwilsdon, 2024-10-18:
It'd be nice to be able to pass around the oneof union used by a message - currently this isn't really possible as the oneof union is defined inside the struct.

My use case is a function like this:

```c
// Subscribe to messages with the provided body.
//
// When a message is received of type which_body, callback is called with its value.
void subscribe(pb_size_t which_body, void (*callback)(Message_body_oneof_t *body));
```

Where `Message_body_oneof_t` is the union.

PetteriAimonen, 2024-10-18:
I wonder if giving the union a name inline with the struct definition is acceptable in standard C99:

    union _MyMessage_myoneof {
        SubMessage oneof_msg1;
        EmptyMessage oneof_msg2;
        SubMessage static_msg;
    } myoneof;

That would be quite trivial change.

jackwilsdon, 2024-10-18:
Yeah that seems to work. Is this something we'd want behind an option or could it just be added (it seems backwards compatible)?

PetteriAimonen, 2024-10-18:
In my opinion it can just be added.

If you can pull-req it, would be great.

This is probably the place: [nanopb_generator.py:1220](https://github.com/nanopb/nanopb/blob/master/generator/nanopb_generator.py#L1220).

I don't think it even needs a test case, though if you feel like it, something like `TEST(sizeof(_MyMessage_myoneof) == sizeof(MyMessage.myoneof))` in tests/oneof should suffice.

jackwilsdon, 2024-10-18:
Opened #1025 - added the sizeof test case you suggested and also a naming test case.

PR #1025 (2024-10-18): Add name to unions
Fixes #1023.
closed 2024-10-18 by PetteriAimonen