nanopb/nanopb #1029: SIGSEV in

KerstinKeller, 2024-10-31:
We've migrated from using the C++ Protobuf implementation to nanopb for the middleware [eCAL](https://github.com/eclipse-ecal/ecal) on our master branch.

So far we haven't had any problems on our dev platforms, but now users reported a SIGSEV thrown in the [`safe_read_bool`](https://github.com/nanopb/nanopb/blob/b36a089ae41284bf4af5230c423750dfbadd649f/pb_encode.c#L115) function.
We're currently using nanopb 0.4.8 (but upgrade would be no problem).

However, we're really at loss, if this is a potential nanopb bug, or if we're doing something wrong trying to encode the protobuf messages.
The call stack goes through `safe_read_bool`, `pb_enc_bool`, `encode_basic_field` `encode_field` and `pb_encode`, which is called from our [encoding function](https://github.com/eclipse-ecal/ecal/blob/c2943a7035e8a0123430c05bfe3ab9e56c6b847c/ecal/core/src/serialization/ecal_serialize_logging.cpp#L72).

What could we be doing wrong that may cause the segmentation fault? Thanks for the help.

PetteriAimonen, 2024-10-31:
That seems quite surprising considering that the message being encoded there doesn't have any `bool` fields.

There was a [recent bug in Firebase](https://github.com/firebase/firebase-ios-sdk/issues/11403) with very similar backtrace, but I think it was very Xcode-specific and was caused by corruption of the `_fields` array.

I do think that corruption of the `_fields` array is the only way `pb_enc_bool` could get called for a message that does not contain it. The type index for `bool` is 0x00, so if the fields array is accidentally zeroed out, it will end up calling `pb_enc_bool`.

If you are able to add logging that would hex dump `eCAL_pb_LogMessage_fields` that could give some clues.
Also if you have more details on the backtrace, such as function arguments or SIGSEGV pointer value, that is helpful.

KerstinKeller, 2024-11-01:
Thanks for your quick answer.
I think it must be some kind of data races, I guess anything else doesn't make sense.

And I am wondering if the nanopb libray itself is threadsafe. Are there any internal states? (in pb_encode / pb_decode)? Do I need to make sure that I don't encode / decode different objects at the same time?

PetteriAimonen, 2024-11-01:
I cannot see any multithreading issue that could cause these symptoms. For `pb_enc_bool()` to get called for that message, the fields array must have been corrupted in some way.

But encoding/decoding different objects is perfectly safe, there is no shared state inside the library.

KerstinKeller, 2024-11-07:
At last we found the issue.
The application that was using our library was also using nanopb, but a much older version.
Hence we had duplicated symbols, in the application and ecal.so, which lead to a corrupt `pb_field_iter_t` in `pb_encode`, returned by an older version of `pb_field_iter_begin`.

Sorry for bothering you!

closed 2024-11-07 by KerstinKeller

PetteriAimonen, 2024-11-07:
No problem, it's interesting to hear about these failure modes.