nanopb/nanopb #1037: Difference between VARINT/UVARINT/SVARINT

inolen, 2024-11-27:
I've been hacking on consolidating varint decoding logic on a side branch, with the most recent commit being -

https://github.com/inolen/nanopb/commit/0634ce74204eff2e7aa5b770e5734b60b5dbdbb7

The general idea is to move the overflow detection into the core decoding routine, so decoding an int*/uint* just involves calling this `pb_dec_varint_n` with the correct size, and decoding an sint* calls the name, with the added zigzag decoding afterwards. I'm doing this so the fast path can directly call this, and then copy off the result without a conditional. My question is, is the only reason for LTYPE_VARINT and LTYPE_UVARINT the bug mentioned at https://github.com/inolen/nanopb/blob/master/pb_decode.c#L1483? It seems that without that bit of logic, they should otherwise be equal. And if that is so, is it still relevant for backwards compatibility, or has enough time passed that they could be consolidated?

PetteriAimonen, 2024-11-27:
Good question.

I think issue #164 is related.

The distinction is also needed for overflow detection. For example `0xffffffffffffffff` can be cast into `int32_t` as value `-1`, but it should not be cast into `uint32_t`.

inolen, 2024-11-27:
Hmm, 0xffffffffffffffff is interesting and I think perhaps the only edge case here.

Anything other than 0xffffffff in the high bits wouldn't look like sign extension.

closed 2024-11-27 by inolen

inolen, 2024-11-27:
@PetteriAimonen i've been trying to reduce the amount of code / overhead that incurs.

Take a look at - https://github.com/inolen/nanopb/blob/codegen2/pb_decode.c#L1497

pb_decode_varint32 and pb_decode_varint now share that code, and the overhead for overflow detection is the popcount in side the byte loop, and a conditional outside of the loop.

PetteriAimonen, 2024-11-27:
@inolen Looks interesting. I'm pretty much falling behind on being able to process these contributions though. Btw, I think the bug you found about pb_decode_delimited and pb_release() is valid, I just haven't had time to look at it either.

inolen, 2024-11-27:
Absolutely no rush, I've just had some free cycles to work on optimizing a side project.