nanopb/nanopb #1039: Nanopb issue with nested object appears to be broken on Big Endian platforms

pdgendt, 2024-11-29:
After bisecting commit 763a2fca0baacd590f32414f143c21ec1646c09d appears to be the first bad commit.

Zephyr runs checks on many different platforms and also on QEMU Malta Big Endian as can be seen here:
https://github.com/zephyrproject-rtos/zephyr/actions/runs/12086504669/job/33705947115

The testing code is located at https://github.com/zephyrproject-rtos/zephyr/tree/main/tests/modules/nanopb

pdgendt, 2024-11-29:
CC @inolen

PetteriAimonen, 2024-11-29:
Hmm yeah, somehow I missed that in review. Taking the first N bytes only works for little-endian.

Looks like it was also caught by the CI, I just hadn't gotten around to looking at the report yet:
https://github.com/nanopb/nanopb/actions/runs/12001876078/job/33453008624#step:4:86

I'm reverting the commit for now.

pdgendt, 2024-11-30:
Thanks for the fast response

closed 2024-11-30 by pdgendt

PetteriAimonen, 2024-12-04:
@inolen Copying first N vs. last N bytes would work if we knew for sure if the architecture is big/little endian, but some platforms don't have the defines. So yeah, I think the size conditional will be needed at least for some platforms but it's possible we could `#ifdef PB_LITTLE_ENDIAN_8BIT` the optimized one. I should really get some reasonable benchmarks working again so it would be easier to judge how this affects things across multiple platforms.

inolen, 2024-12-04:
Got it. I'll rebase my codegen branch against latest and revert the memcpy.