nanopb/nanopb #1041: Automate release workflow, switch to GitHub releases

PetteriAimonen, 2024-12-01:
Nanopb has been using custom hosted download server as back when it was set up, GitHub didn't have releases functionality yet.

Nowadays it would simplify release management to have them as GitHub releases instead.

tpudlik, 2024-12-16:
As part of the release workflow, could [the version in MODULE.bazel](https://github.com/nanopb/nanopb/blob/master/MODULE.bazel#L3) be incremented? I noticed this was not done for the 0.4.9.1 release. I patched this line when adding the release to the Bazel Central Registry, but it would be nice for it to be correct in the source.

PetteriAimonen, 2024-12-17:
Yeah, that should be added to `set_version.sh`: https://github.com/nanopb/nanopb/blob/master/tools/set_version.sh

weichslgartner, 2025-01-10:
Could the version information also be added to `zephyr/module.yml` (see https://docs.zephyrproject.org/latest/develop/modules.html#vulnerability-monitoring). This would enable to generate SBOMs in zephyr projects using nanopb.

For example adding CPE and PURL for version 0.4.0 would look like:
```yml
security:
  external-references:
    - cpe:2.3:a:nanopb_project:nanopb:0.4.0:*:*:*:*:*:*:*
    - pkg:github/nanopb/nanopb@V0.4.0
```

PetteriAimonen, 2025-01-10:
@weichslgartner Sure, can you make a pull request for set_version.sh modification?

weichslgartner, 2025-01-10:
@PetteriAimonen Sure, will create an PR.

pdgendt, 2025-01-17:
Any ETA on the 1.0.0 release?

PetteriAimonen, 2025-01-17:
Probably many months, I haven't done much towards it.