nanopb/nanopb #1043: The conditional judgment in the pb_write function is not necessary

LittleFatHero, 2024-12-06:
**file:** pb_encode.c

there is a conditional judgment in the function pb_write as follow:
```
if (stream->bytes_written + count < stream->bytes_written ||
            stream->bytes_written + count > stream->max_size)
        {
            PB_RETURN_ERROR(stream, "stream full");
        }
```
I think "stream->bytes_written + count < stream->bytes_written" is not necessary, because count > 0, it is always true.
maybe it is better to write as follow：
```
if (stream->bytes_written + count > stream->max_size)
        {
            PB_RETURN_ERROR(stream, "stream full");
        }
```
it can't pass the coverity code scan system if "stream->bytes_written + count < stream->bytes_written" is there.
if using coverity to scan the code, there will be a error if "stream->bytes_written + count < stream->bytes_written" in the code, because
coverity think that it is always true and it is not necessary!

PetteriAimonen, 2024-12-06:
It checks against unsigned integer overflow (for example if bytes_written = SIZE_MAX/2 and count = SIZE_MAX/2+1). Because unsigned overflow is fully defined in C11 and later, I'd expect code analysis systems to understand it.

There may be an alternative formula that is equivalent and I'm ok with replacing it, but I think the overflow check is necessary in one form or another.

LittleFatHero, 2024-12-06:
Got it! Thank you so much!

closed 2024-12-06 by LittleFatHero