nanopb/nanopb #1044: Does nanopb support running in 64-bit cortex-R or cortex-A microcontrollers？

LittleFatHero, 2024-12-06:
Hi， 
Does nanopb support running in 64-bit cortex-R or cortex-A microcontrollers？
Is there any jobs I need to to in order to run nanopb in 64-bit cortex-R or cortex-A microcontrollers?

PetteriAimonen, 2024-12-06:
> References

Hi,

Sure. It's pretty portable to all common architectures.
Probably about a billion 64-bit Cortex-A systems running nanopb.

LittleFatHero, 2024-12-09:
> Hi, Sure. It's pretty portable to all common architectures. Probably about a billion 64-bit Cortex-A systems running nanopb.
> […](#)
> -- Petteri

Got it!  Thank you so much!  By the way, 
1) Is there any different between running nanopb on a 32-bit system and a 64-bit system? 
2) Is there any macro definitions related to 64-bit systems do I need to enable?

PetteriAimonen, 2024-12-09:
No, it should just work.

LittleFatHero, 2024-12-09:
> No, it should just work.

Got it!

closed 2024-12-09 by LittleFatHero