nanopb/nanopb #1046: Support Bazel 8 in CI

tpudlik, 2024-12-16:
Currently, `bazelisk test //...` fails on Bazel 8.0.0: https://github.com/nanopb/nanopb/actions/runs/12344585119/job/34447334503

The error is caused by `extra/bazel/nanopb_cc_proto_library.bzl` pulling in a toolchain that no longer exists in the new Bazel version.

I dug into this a little bit, and the problem appears to be that `toolchains_protoc` (a transitive dependency pulled in via `rules_proto_grpc`) is not yet compatible with `rules_proto@7.0.2`, which Bazel 8 ends up pulling in.

terry-sungbae-yoon, 2025-02-05:
Huge thanks for summarizing this issue. Helped me save a ton of time.

`toolchains_protoc`[@0.3.7](https://registry.bazel.build/modules/toolchains_protoc/0.3.7) is now released with support for bazel 8.

Declaring it an explicit dependency of my project which imports `nanopb`, I was able to get the same build result with bazel 8 when compared to bazel 7. Yet, I haven't examined if such change in `rules_proto_grpc`'s dependency comes with possible side effects.

schultetwin, 2026-09-03:
I believe this is fixed here: https://github.com/nanopb/nanopb/commit/9a5ec7131de9d89e0bf85116e9eafeb8b5734a1f

PetteriAimonen, 2026-09-03:
Yeah. If I understand correctly, that update can be breaking change for some users, so I'm not planning to put it into 0.4.9.x maintenance releases, but rather to 1.0 which I still hope to get out this year.

schultetwin, 2026-09-03:
👍 Got it, thank you! Yes, definitely a breaking change for Bazel users on Bazel 7.