nanopb/nanopb #1049: Decoding complex messages with multiple structs

ramyezzatmapzon, 2025-01-09:
Hi, 

Trying to decode complex messages, and keep getting the following errors 

```
E (533) Protobuf: Decoding failed: zero tag 
E (533) Protobuf: Decoding failed: invalid wire_type 

```

message.proto 
``` 
syntax = "proto3";   

message DeviceStatusEvent {
  int32 battery_soc = 1; 
} 

message StatusMessage { 
  int32 input_voltage = 1; 
  uint32 battery_soc = 2; 
  int32 battery_temperature = 3; 
  uint32 timestamp_ms = 4; 
  string imei = 5; 
  string fw_version = 6; 
  repeated DeviceStatusEvent device_events = 7; 
} 
``` 

the generated files are as follwos
```
/* Struct definitions */
typedef struct _protocol_gateway_DeviceStatusEvent {
    int32_t battery_soc;
} _protocol_gateway_DeviceStatusEvent;

typedef struct _protocol_gateway_StatusMessage {
    int32_t input_voltage;
    uint32_t battery_soc;
    int32_t battery_temperature;
    uint32_t timestamp_ms;
    pb_callback_t imei;
    pb_callback_t fw_version;
    pb_callback_t device_events;
} protocol_gateway_StatusMessage;
```

parent message callback assignation from decoding function 

``` 
mapit_protocol_gateway_DeviceStatusEvent events[10] = {0}; 
message.device_events.funcs.decode = decode_device_event_callback; 
message.device_events.arg = events; 

``` 

callback function  
``` 
// Callback function to provide string data during encoding 
bool encode_device_event_callback(pb_ostream_t *stream, const pb_field_t *field, void *const *arg) { 
    protocol_gateway_DeviceStatusEvent *events = (const protocol_gateway_DeviceStatusEvent *)(*arg); 
    for (size_t i = 0; i < 10; i++) { 
        if (!pb_encode_tag_for_field(stream, field)) 
            return false; 
        if (!pb_encode_submessage(stream, protocol_gateway_DeviceStatusEvent_fields, &events[i])) 
            return false; 
    } 
    return true; 
} 
``` 

Encoded message: 

08 B0 09 10 55 18 23 20 C5 FF 96 AB 06 2A 0F 31 32 33 34 35 36 37 38 39 31 32 33 34 35 36 32 08 46 57 5F 31 2E 30 2E 30 3A 02 08 50 3A 02 08 4B 3A 02 08 32 3A 00 3A 00 3A 00 3A 00 3A 00 3A 00 3A 00 
  

The message seems to be decodable from online protobuf decoder, so i am assuming i am doing something wrong. Please note that i am not using pb .options nor dynamic memory

[image]

PetteriAimonen, 2025-01-09:
It seems that you have included the encoding callback function, while asking about decoding.

You could try specifying `[(nanopb).max_size = 10]` for the array to see if the message can be decoded when not using callbacks. This would inform if the problem is within your custom callback function.

ramyezzatmapzon, 2025-01-09:
Updated the function to decoding function in the issue description

Will try to add the max_size option, for some reason the generator cannot locate the "nanopb.proto"

PetteriAimonen, 2025-01-09:
Regarding the decode callback, it will get called repeatedly and each time there will be just one submessage to decode. But as written it shouldn't give an error, it would just always overwrite the first array entry.

Check that the total length of the message that you give to `pb_istream_from_buffer()` is correct (and is the size of the message, not the full buffer).

Regarding generator options, you can also use a .options file (which won't need nanopb.proto import).

ramyezzatmapzon, 2025-01-10:
Thanks a lot for your support, the length of message that is given to `pb_istream_from_buffer()` had to match the message size

closed 2025-01-10 by ramyezzatmapzon