nanopb/nanopb #1053: Support linking in multiple different nanopb versions

tnek, 2025-01-22:
I'm attempting to build a target that statically links nanopb 0.4.9.1 with multiple other dependencies that also depend on and dynamically different nanopb versions. As-is, all of the nanopb definitions occupy the global namespace, resulting in surprising behavior during encode/decode depending on link order

This is especially a problem for dependencies with nanopb < 0.4.3 and nanopb > 0.4.3, as that version reorders the fields within pb_msgdesc_t, leading to memory corruption on decode/encode. 

One way this can be accomplished is by allowing for a `--cpp-namespace`option that would encapsulate the emitted protoc output in a given namespace.

PetteriAimonen, 2025-01-23:
Could this perhaps be done in your build scripts, by using `objcopy --prefix-symbols`? Assuming you can separate the different version usages into different intermediate `.o` object files?

KammutierSpule, 2025-03-25:
I'm facing this issue, would a wrapper around the files work?

```
nanopb_version.cpp
namespace nanopb_version {
#include "nanopb.c"
}

compile_3rdparty_library.cpp
using namespace nanopb_version;
#include "3rdparty_library.c"

```