nanopb/nanopb #1055: Options files not found in subfolders if NANOPB_GENERATE_CPP_STANDALONE is set to OFF

axelarbona, 2025-02-04:
Hello,

I am trying to use `.options` files in subfolders with `NANOPB_GENERATE_CPP_STANDALONE` set to `OFF` but then none of my `.options` files are being found. If I do not set `NANOPB_GENERATE_CPP_STANDALONE` this does not happen and my `.options` are being used correctly, but the output files do not follow the original folder structure.

The tree looks like this (and files have references between each other):

```
protos
├── folder_0
│   ├── a.proto
│   └── a.options
├── folder_1
│   ├── b.proto
│   ├── b.options
│   ├── c.proto
│   └── c.options
...
```

If I set `set(NANOPB_GENERATE_CPP_STANDALONE OFF)` then none of the options are being used and I get a warning about the missing files.
I tried including them manually in `NANOPB_DEPENDS` or by adding the paths directly in `PROTOC_OPTIONS` with `--nanopb_opt=-I <path>` with no success (it always says the file can't be found).

Looking at the [FindNanopb.cmake](https://github.com/nanopb/nanopb/blob/master/extra/FindNanopb.cmake#L28) comment, I do expect the output to be in a tree or flat depending on that variable, but not the change in finding the options files. Is this a bug that can be fixed? Or is there any way around this?

One workaround I could think of is to concatenate all my options files into one single gigantic file, and then pass that as a `-f` parameter, but that feels like a huge hack, and I get a big warning for each proto being compiled for each option that does not match it... I will try that out just in case in the meantime.

Thank you very much in advance.

PetteriAimonen, 2025-02-04:
That seems very surprising, as I don't see anything in the file that should make a difference for this.

Something weird is happening for me too, as it seems I got the problem to occur once, but then after removing the build folder and rerunning cmake from clean, it didn't reoccur.

It could be useful to check `make VERBOSE=1` output to see what command it is running for protoc.

axelarbona, 2025-02-04:
Thank you for your fast reply. I will keep debugging this today and see if I can find something.

Our build system is fairly complex, with a lot of moving parts involving cmake, earthly, and a lot of very big proto files, so debugging is pretty annoying. However, as we do use earthly, I would expect all builds to start from clean. I will try to generate a simpler study case and see if the bug still happens there. Hopefully with that I can better understand what is happening and ease debugging.

I was hoping for a "ah, you forgot to do this and that's why it is not working!" but I guess it was not that simple :(

closed 2025-02-05 by axelarbona