nanopb/nanopb #1057: Request for Privacy Manifest Compliance for Nanopb Framework

Aamir369, 2025-02-08:
Dear nanopb team,

I hope this message finds you well.

I am currently developing an iOS application that includes the nanopb.framework. Upon submitting the app to the Apple App Store, I encountered an issue flagged by Apple (ITMS-91061: Missing Privacy Manifest). Apple now requires that third-party SDKs include a privacy manifest file outlining data collection practices, effective February 12, 2025.

I wanted to inquire whether there is an updated version of the nanopb.framework that complies with this requirement. If not, do you have plans to release a compliant version with a privacy manifest file soon? This would be crucial for ensuring continued use of your framework in iOS applications.

Thank you for your hard work and support for the development community. I look forward to your response.

Best regards,
Aamir Baig Mughal
aamir.mughal@kineticdataminds.ca

PetteriAimonen, 2025-02-09:
I have the understanding that this was added in commit 00227bd69 by @ncooke3.

Myself I don't do any Apple development, so I'm unable to help with this issue.

ncooke3, 2025-02-10:
Hi @Aamir369, what version of nanopb is your application using? Are you using Firebase as well? And, how are you integrating nanopb (CocoaPods, SPM, etc.)?

PetteriAimonen, 2026-01-30:
I'm closing this issue as I don't know what it even means.