nanopb/nanopb #1058: Message is emty (has zero length) if all fields are zero

andrey-pr, 2025-02-09:
Hello.
I encountered problem with encoding message of zero values: nanopb does not recognize it as message to encode and returns nothing in output.
Current message contains two fields: int "error" and bytes "read". When I set "error" to zero and "read" empty, I got answer_length of zero, but normally empty message still have size of several bytes. How can avoid this problem?
Thank you. Nanopb is helped me a lot in my project

```c++
SPIReadWriteResponse answer = SPIReadWriteResponse_init_zero;

pb_ostream_t stream2 = pb_ostream_from_buffer(buffer, sizeof(buffer));
        
answer.error = 0;
answer.read.size = 0;
memcpy(answer.read.bytes, readed_value, message.rn_bytes);
        
status = pb_encode(&stream2, SPIReadWriteResponse_fields, &answer);
size_t answer_length = stream2.bytes_written;
```

PetteriAimonen, 2025-02-09:
AFAIK this is the correct encoding of a message that has no non-zero fields in proto3 format. Have you tried whether the same occurs in other protobuf implementations with the same message? Protoc seems to agree here:

```
$ cat test.proto
syntax = 'proto3';
message MyMessage {
	int32 error = 1;
	bytes msg = 2;
}
$ echo -e 'error: 1234\n msg: "abc"' | protoc --encode=MyMessage test.proto | xd -p
     0: 08 D2 09 12 03 61 62 63
$ echo -e 'error: 0\n msg: ""' | protoc --encode=MyMessage test.proto | xd -p
$ 

```

If the zero length message is a problem for your transport format, you could add a dummy field that you always set to a non-zero value. Making the proto3 fields `optional` also makes them explicitly present even when zero.

andrey-pr, 2025-02-09:
> AFAIK this is the correct encoding of a message that has no non-zero fields in proto3 format. Have you tried whether the same occurs in other protobuf implementations with the same message? Protoc [...]

Thank you for quick answer. According to https://www.protobufpal.com/, this message should be encoded to [0x08, 0x00, 0x12, 0x00]. If all fields of message are zero, it does not means that has no sense, so it have to be encoded in some way

PetteriAimonen, 2025-02-09:
Zero bytes long is a valid protobuf message, so it is encoded.

The encoding `0x08, 0x00, 0x12, 0x00` encodes zero values, and does not appear to follow the proto3 implicit field presence rules. Note that if you input empty hex message to protobufpal and press decode, you get the same output.

Nanopb follows the protobuf specification as implemented by Google's own protobuf libraries. It appears that protobufpal does not, which is ok as far as interoperability goes, but I don't see a need to deviate here.