nanopb/nanopb #1059: Request: Generate the expected storage size on repeated only messages

KammutierSpule, 2025-02-12:
Consider:
```
message s_message
{
  repeated s_message1 message1 = 1 [ (nanopb).type = FT_POINTER ];
  repeated s_message2 message2 = 2 [ (nanopb).type = FT_POINTER ];
}
```

This proto does not generate a  `#define s_message_size               ...` so I don't know the minimal expected "overhead size" of storing these arrays.

Note I understand that all eventually final size of the max encoded depends on s_message1 and s_message2.. but isn't there a size overhead for this message?

PetteriAimonen, 2025-02-12:
The size overhead will also depend on the length of the arrays.
Typically it is 1 byte for tag + a few bytes for length + then the submessage data.

Nanopb only generates the define when it knows the maximum size of the message, including all of its contents.

KammutierSpule, 2025-02-12:
I found this on the generated code:
```
/* Maximum encoded size of messages (where known) */
/* s_message_size depends on runtime parameters */
```
but even if I add a fixed 
```
message s_message
{
  repeated s_message1 message1 = 1 [ (nanopb).type = FT_POINTER ];
  repeated s_message2 message2 = 2 [ (nanopb).type = FT_POINTER ];
  required bool dummy = 3;
  ...
}
```
It wont' give any hint if I keep adding elements.
Would be benefit to have these figures?

> The size overhead will also depend on the length of the arrays.
> Typically it is 1 byte for tag + a few bytes for length + then the submessage data for each item in the array.

So like the other s_message_size sizes, the maximum possible "1 byte for tag + a few bytes for length" is possible to know?

PetteriAimonen, 2025-02-12:
I don't see any benefit for a value that could be less or more than the actual message size. Without more information the generator cannot calculate minimum nor maximum.

You can get actual message size by `pb_get_encoded_size()` at runtime.

You may know more about your data than nanopb does, in which case you can calculate the size based on protobuf encoding format specification.

KammutierSpule, 2025-02-12:
Usually I allocate the maximum size in advance:

```
uint8_t* data_nanopb_encoded = new uint8_t[s_message_size]; // allocate the maximum expected size
pb_ostream_t stream = pb_ostream_from_buffer((pb_byte_t*)data_nanopb_encoded, s_message_size);
pb_encode(&stream, s_message_fields, message_structure)
```

I may consider pb_get_encoded_size, but it is a CPU time over RAM tradeoff.

KammutierSpule, 2025-02-13:
If I understand correctly from:
https://protobuf.dev/programming-guides/encoding/#optional

> Ordinary (not [packed](https://protobuf.dev/programming-guides/encoding/#packed)) repeated fields emit one record for every element of the field.

This means that I just need to know the (static) size of the message (which nanopb give us, and has already the overhead per encoded message size) and multiply by the number of maximum expected messages (which I know).

closed 2025-02-13 by KammutierSpule