nanopb/nanopb #1061: Decoding submsg_callback with oneof FT_CALLBACK fails silently

HarshaLaxman, 2025-02-24:


PetteriAimonen, 2025-02-24:
Yeah, this is a common problem people run into, for which there are various workarounds but no universal solution.

From [docs](https://jpa.kapsi.fi/nanopb/docs/concepts.html#oneof):
> When a field inside oneof contains pb_callback_t fields, the callback values cannot be set before decoding. This is because the different fields share the same storage space in C union. Instead either function name bound callbacks or a separate message level callback can be used. See [tests/oneof_callback](https://github.com/nanopb/nanopb/tree/master/tests/oneof_callback) for an example on this.

You can also consider using `no_union` option. Or of course, if you know the maximum string length, just set it to the generator `max_size` option and avoid callbacks altogether.

PetteriAimonen, 2025-02-24:
Ah, sorry, I didn't read your code carefully enough.

Indeed this feels like an omission, as currently the submessage callback is only invoked when the submessage itself is static, and callbacks are only used inside of it. I guess it would be possible to copy the logic from `pb_enc_submessage` / `pb_dec_submessage` to `pb_default_field_callback`.

Can you elaborate on the issues you are having with name-bound callbacks? I hope that eventually I would be able to come up with a better solution for this long standing issue, so I'm trying to understand the practical problems.

HarshaLaxman, 2025-02-24:
Name bound callbacks feel shaky

For example, iiuc in the [tests/oneof_callback](https://github.com/nanopb/nanopb/tree/master/tests/oneof_callback), adding another message would cause the `SubMsg3_callback` to be ran for `AnotherMessage` also, with no additional code.
```
message AnotherMessage
{
    option (nanopb_msgopt).submsg_callback = true;

    oneof values
    {
        int32 intvalue = 1;
        SubMsg3 submsg3 = 2;
    }
}
```

Even if there is a way to add code to check which top level message the field belongs to in `SubMsg3_callback`, it's a bit of hidden logic.

It's preferable to explicitly set the callbacks from the top level message on down.

HarshaLaxman, 2025-02-25:
Put up a PR with a fix and regression test. All tests pass, but it's likely I missed some edge cases

https://github.com/nanopb/nanopb/pull/1063/files