nanopb/nanopb #1064: Custom naming with prefix depending on type

AlexanderVonGlowacki, 2025-02-27:
The custom naming feature from #1026  could be changed so that the type of the field is also transferred in addition to the name.
 
For some coding guidelines it is necessary to add a type-dependent prefix to the variable name.

PetteriAimonen, 2025-02-27:
Sounds reasonable, though I'm not going to work on this myself.

@joshstrohminger Just a ping, in case you have any thoughts.

joshstrohminger, 2025-03-02:
@AlexanderVonGlowacki, I suppose it depends on how specific your notation needs to be, and therefore how much type information and context is required.

If you want to differentiate between different sizes, signs, pointers vs values, callbacks, or repeated vs singular, some of that information is available from some callers of the method `var_name(...)`, such as fields and field lists. You can pass their `cytpe`, `pbtype`, `allocation`, and `rules` attributes to repeat some of the logic in https://github.com/nanopb/nanopb/blob/8e9d5a10125b596188e95e925937d380a3c79e8e/generator/nanopb_generator.py#L759-L791
But not all callers have access to this level of information, mostly because it doesn't quite apply, like unions/oneofs and extensions. You might need some way to indicate what those are, perhaps passing something to indicate the context of the type information. If you didn't need this level of notation and only wanted to differentiate between basic data types, maybe just the `ctype` could be enough.

There is also the matter of the "built-in" fields with prefixes/suffixes like `has_`, `which_`, `_count`, and `_default` (maybe some others I missed) that are applied after applying the naming style. When I added the custom-naming feature I left these alone because I figured it was better that these stand out as being "meta" fields and not true fields of the message. I was also afraid of overstepping. They could be included in styling but maybe they'd need some more specialized methods added to the `NamingStyle` classes in case people want to differentiate them.

Then you'd need to decide whether you want to introduce what I would consider to be a breaking change by adding additional arguments to the `var_name` method. If not, you could patch the custom styler's `var_name` method if it has too few arguments. Something like the following, assuming you need to add two more args (I have no idea if this is a reliable means to doing this):
```python
if Globals.naming_style.var_name.__code__.co_argcount == 2:  # self and name
    original_method = Globals.naming_style.var_name
    def patched_method(name, new_arg_1, new_arg_2):
        return original_method(name)
    Globals.naming_style.var_name = patched_method
```

AlexanderVonGlowacki, 2025-03-03:
Thank you for your response @joshstrohminger .
I mainly want to distinguish between the different data types and whether it is a pointer or an array (pointer). The information from the caller of the `var_name` method should be enough. The output could look like this:
`uint32_t u32MyUnsignedLongVar;
uint16_t *pu8MyUnsignedShortPointer;
float *(*papfArray)[5];`

> There is also the matter of the "built-in" fields with prefixes/suffixes like has_, which_, _count, and _default
I would not touch these fields either. It is far easier to stick with the familiar naming here.

Can you consider a change, or should I modifying my generator myself, perhaps with something like function overloading (altough python does not nativly support this)? However, this could mean additional work and potential problems for future updates.

joshstrohminger, 2025-03-06:
This is how I was envisioning handling the overloading problem, assuming you had expanded the signature of `var_name` to include the extra fields:

```python
Globals.naming_style = InheritNamingStyle()

if Globals.naming_style.var_name.__code__.co_argcount == 2:  # self and name
    original_var_name = Globals.naming_style.var_name
    
    def patched_var_name(name, ctype='', pbtype='', context=''):
        return original_var_name(name)
    
    Globals.naming_style.var_name = patched_var_name
```

Of course, this may not be necessary. Reading https://github.com/nanopb/nanopb/issues/1066, you could potentially just include this as a breaking change for a 2.x release with the new method signature for `var_name` and no patching/overloading. Or maybe even a 1.x release if it gets done in time since it hasn't actually been tagged/released yet.

AlexanderVonGlowacki, 2025-03-06:
Is it possible that you could add this function using the same mechanism as you did with the other custom naming function? It seems you already have an idea, and I'm not that familiar with Python. What would be the effort involved?

joshstrohminger, 2025-03-13:
The effort is hard to say. It might be low, just update the signature and all of the callers in _nanopb_generator.py_, and add some tests. But if you stumble onto something that doesn't work as expected (like I did when adding the custom naming feature), especially once adding testing, it could balloon. I don't have a lot of time right now, but if I find some, I'll try to put a quick change together in my fork that you can experiment with and provided feedback.

AlexanderVonGlowacki, 2025-03-14:
Thank you, i will do so.