nanopb/nanopb #1066: Review of Nanopb 1.x API plans

PetteriAimonen, 2025-03-05:
The next stable series of nanopb will be 1.x. The project is switching to semantic versioning system, so the API will remain stable until 2.x. This is a chance to improve upon the 0.4.x API.

I would like feedback from the contributors and users on the API decisions.

Here is a ping to recent contributors that could have an interest in these.
If you have time and interest to review and comment, that's great, but I totally understand if not :)
@joshstrohminger @pdgendt @zachdeibert @tpudlik @mark64 @jaskij @ConradWood @HarshaLaxman @twam @Sayter99 @denravonska @inolen @gsurkov

## Improve, or release fast?
I could release mostly unchanged 0.4.9 version with small improvements as 1.0.
Alternatively I can delay 1.0 release until the improved API is done.

In any case I would eventually want to get the API changes done, so that would be 2.0 then.
Having more stable series would increase maintenance burden, as critical bugfixes would need to be backported to at least 1.x and 0.4.x series. I do aim to backport things back to 0.1.x still, if the bugfix is critical enough.

## API plans
Regardless of whether it's going to be 1.0 or 2.0, here are my preliminary plans for API changes I am planning:

### Context structures
Replace `pb_istream_t` and `pb_ostream_t` stream structures with `pb_decode_ctx_t` and `pb_encode_ctx_t` context structures. The context structures would contain more than what streams currently do:

* Per-context shared callback function. This would have a prototype `bool field_callback(pb_decode_ctx_t *ctx, const pb_field_iter_t *field)`. The callback can then determine through field descriptor what it should do, i.e. what field of what message is being processed. The purpose of this change is to keep function pointers out of the message structure (which causes problems with oneofs and is also potential security risk if something goes wrong). Unlike name-bound callbacks of 0.4, it would permit handling the callbacks differently depending on situation. This would help issues like #1061, #884, #572.
* Per-context allocation arena. Memory is practically always allocated sequentially in nanopb, only growing the latest allocation. As mentioned in #571, on many platforms `realloc()` is actually ineffective at growing allocations. By having our own arena per context, we can efficiently grow the latest decoded repeated field. Using arena allows easily to restrict the amount of memory that can be used for the decoding.
* Per-context optional `realloc()` function. This would be called when the arena size runs out. If no realloc function is given, it acts like currently when dynamic allocation is disabled. The function would be passed the context, resolving #779.
* Context structure would always be passed as the original pointer. Unlike the substreams used currently, this would allow extending the context structure to include user data (#776).

### Reconsider default fallback field type
Currently the generator falls back to callback type when field size is not known.
This tends to confuse new users.

Especially with the arena allocation, `FT_POINTER` would make more sense as the fallback type.
If the user does not give an arena, this would then fail with an error message that would hopefully make it more clear what is happening.

Another alternative would be to generate statically allocated fields where the size would be a define such as `MYMESSAGE_MYFIELD_MAXSIZE`.
Then there would be a `#ifndef MYMESSAGE_MYFIELD_MAXSIZE` `#define MYMESSAGE_MYFIELD_MAXSIZE PB_DEFAULT_FIELD_MAXSIZE` or similar generated. This could make it easier to understand the maximum sizes. This is related to what was proposed in #707.

In any case the fallback type could be overridden with the existing generator option `fallback_type`.

On the other hand, this is partially just a documentation issue. Better tutorials could resolve it.

### Improve bytes field pointer type
Currently bytes fields with `FT_POINTER` are generated as a pointer to a structure with size and data.
In many cases it would be more useful to have the size, and then a pointer to the data.

For repeated fields where both the array and the bytes sizes are dynamic, it would be a pointer to an array of structures, where each has the size and the pointer to data.

### Memory buffers as the primary I/O mechanism
Currently nanopb first and foremost has callback-based I/O streams. This is related to my hyper-optimization of memory usage back when I created nanopb.

Overwhelming majority of nanopb users decode and encode to memory buffers.
The current structure of the code causes some overhead here, though `PB_BUFFER_ONLY` reduces the overhead.

An alternative would be to have the context include a pointer to an input/output memory buffer.
Data would be transferred directly from it.

To support the case of streams, the context would include a function pointer for a callback that refills the input buffer or writes out the output buffer.
If the buffer is larger than 1 byte, it acts as a cache to reduce number of I/O requests.
Items larger than the buffer size would be transferred in a single callback call.
Setting the buffer size to 0 bytes would restore current behavior of callback per each internal read/write.

### Possibly rework the field descriptor format
The field descriptor format introduced in 0.4 has a few downsides:

* Not as flexible as I hoped, adding new features is still often made difficult for not having space for new descriptor information.
* The macro mess that generates it is horrible. (I do like the generated field list in `myproto.pb.h`, but would like to simplify the macro logic in `PB_BIND()`)
* Somewhat efficient in ROM usage, but slow to access.

I'm considering trying to change the encoding to improve access speed, even if it slightly increases per-field ROM usage.
I'm not yet sure how exactly to implement this.
Especially keeping support for AVR progmem storage is difficult.

### API and ABI version compatiblity
I plan to introduce a `PB_API_VERSION` define. When not given, the API would include only latest, non-legacy functions. When `PB_API_VERSION` is defined to e.g. `0` for `0.x`, compatibility functions would be defined to simplify porting from previous versions.

There have been interest in ABI compatibility between nanopb versions.
First and foremost, this would require eliminating compilation options that affect the ABI.
This would include at least `PB_FIELD_32BIT` and `PB_ENABLE_MALLOC`.
This could be done so that defaults are full-featured, and then specific disable macros would explicitly cause incompatible ABI.

pdgendt, 2025-03-05:
Thanks for gathering all upcoming API changes here, I think these make sense from a high-level point of view.

Regarding the version number, I don't really have a strong opinion, and I think users are more interested in a timeline instead. One remark for Zephyr specifically, is the following
https://github.com/nanopb/nanopb/blob/98bd557624c227373633e3c81ece1a73990b3939/zephyr/module.yml#L11-L12
already refers to `1.0.0` while not being released yet. So I think we jumped the gun in #1050.
This hasn't been updated yet in Zephyr, so we could update the tag on another release.

> Context structures

I approve the usage of context structures over streams, and the flexibility it brings, looking forward to that.

> Reconsider default fallback field type

For the default callback, I agree that documentation can already prove sufficient, I would consider this a lower priority.

> Improve bytes field pointer type

Yes, please :)

> Memory buffers as the primary I/O mechanism

I tend to use both strategies, depending on the use case, but prioritising memory buffers makes sense.

> Possibly rework the field descriptor format

Not sure how visible this is for users, would need to dive into the implementation details.

> API and ABI version compatibility

+1

ConradWood, 2025-03-05:
First of, thank you for spending time listing the plans and thinking of
my usecase (the context). Much appreciated!

To answer the "Should I wait or release fast" question:
I prefer the "wait".
This is because, nanopb currently "works for me" and upgrading and
rolling it out to devices in-the wild is quite a task. Our devices have
combinations of binary firmware blobs, so ensuring compatibility with
existing code is tricky. I would probably skip 0.4.9 for that reason.

Context structures:
That is a very fine feature and I am happy to add some code for it once
we know more how it should look like.

ABI compatibility:
Yes, I am keen on that. Sometimes though I think sacrificing ABI
compatibility for new features is unavoidable.
If that happens, it is most important that it can be detected in code.
It sounds like PB_API_VERSION is meant to do that.
Thank you, that's great!

Another issue I am keen on:
Background: I run a proto repository server which keeps .proto files
and their compiled outputs in various languages available.
(Currently golang, nanopb, python, php, java).
It's a bit fiddly at times.
What helps is:
* that generated filenames are predictable
* that any generated filenames (and respective includes) are in
subdirectories, e.g.
  [protopackage]/myproto_def.h
  [protopackage]/myproto_pb.c
  [protopackage]/myproto_pb.h
* that nanopb itself is in a subdirectory (perhaps optional as compiler
option), e.g. #include <nanopb.pb.b>
   (this makes updating nanopb simple and scriptable - just remove
files in that dir and copy new ones into it)

  Note: I referred to [protopackage] above.
  The  "package" definition in protos is a bit ambigous, we have
"package statement" and "option go_package" and "option java_package"
and possibly others.
  It seems to me the intended usage this system is for proto compilers
to add an option, for example, "nanopb_package". The Compiler could
derive  a target directory from that.
  (go and java do it like that)
  Sorry to add to the list of things, but it does seem the right time
to do it, if we're considering API and ABI changes.

Other items:
I don't feel I can comment - I have no strong opinion on any of those.

Once more, thanks for your efforts!

Conrad

jaskij, 2025-03-06:
Fast or wait?

I'd wait.

0.4.9 is good as is, and if you release 1.0 with it as-is,
then you're releasing it with a deadline. We'll have 1.0 now, and 2.0 in a year or so,
because the breaking improvements are too important.

Just about the only thing about 1.0 would be impressing the people who think that
anything starting with a 0 is an unstable, not production ready version. So it could,
maybe, increase adoption of nanopb.

But then again, it's 1.0 now, and then 2.0 in a year or two because you need to break the API
to improve the library.

----

# API changes

## Reconsider default fallback field type

I like the define better, but instead of defaulting to `PB_DEFAULT_FIELD_MAXSIZE`, which
I assume could be large, and throw linker errors, I'd just stick an `#error "Please define ..."`
in there.

## Improve bytes field pointer type

I like the change. One less pointer to chase, and just seems friendlier.

## Memory buffers as the primary I/O mechanism

Agreed. 

## Description format rework

I'd say go for it. Even if it's a slight increase in size, if it's easier for you, faster,
and more future proof at the cost of, say, 10% more ROM usage only for descriptos?
Yeah, that's totally fair.

I genuinely thought AVRs were dead outside of hobby projects. Turns out, Microchip released
the latest AVR MCU in 2020. Color me surprised.

Personally I have strong opinions about the continued existence of 8-bit MCUs. There probably will
be some strong disagreement from someone, but between 8-bit being (too) slowly phased out and
the big maintenance burden on you, my suggestions would be:

- drop AVR (and 8-bit in general) support in 1.0
- accept bugfixes for 0.4 as a form of support for AVRs

I'm hoping that keeping around the old version in a bugfix-only status will be a lower load
that trying to keep the support alive in 1.0.

## API and ABI compatibility

Semver for API is nice, at least everyone understands it at a glance.

ABI stability is... difficult, especially for the C++ bindings. It's absolutely doable, just difficult.
Perhaps do a best-effort ABI stability thing, encourage users to open bug reports whenever the ABI
breaks and see how many actually care enough to open an issue?

----

@ConradWood

For generating polyglot outputs from `.proto` files, the [`buf` CLI](https://buf.build/)
is an absolute godsent, even if you do not use their SaaS registry offering.

mark64, 2025-03-06:
> ## Improve, or release fast?
> I could release mostly unchanged 0.4.9 version with small improvements as 1.0. Alternatively I can delay 1.0 release until the improved API is done.
This is really a question for you on what you want to maintain. No opinion from me. Probably easier to hold off on 1.0 for your own sanity.

> ### Context structures
Seems reasonable

> ### Reconsider default fallback field type
No opinion as it doesn't affect any of the places I've worked at, all our fields were intentionally chosen to be static sizes.

> ### Improve bytes field pointer type
Similar to above, no opinion

> ### Memory buffers as the primary I/O mechanism
This is good, and maybe I'd go further. One use case I had was unable to use nanopb due to slow decode and encode performance. Of course the buffer only define fixes that somewhat, but it wasn't enough. I'd be worried that this caching implementation might end up hurting performance.

From the use cases I managed, a hyper-optimized memory buffer version would be highly preferable. Combined with possible optimizations to the field descriptor you mentioned below, that would be great for high-performance embedded projects.

I'd recommend considering callback IO secondary, possibly even supporting it via a different implementation.

> ### Possibly rework the field descriptor format
While some places I know are optimizing for all the ROM space they can get, that may be a case where 0.4.x would have to be used. Reworking the format for speed, even at a 50% RAM size increase (or maybe even more) would be beneficial for my uses.

Finally, I'd also consider ways to reduce the memory usage and CPU overhead of parsing recursive messages or unions/oneofs. Our performance noticeably tanked when we split out large messages into sub-messages, but putting all the data into a single message was quite messy.

twam, 2025-03-07:
No need to hurry for a 1.0 version bump from our side.

Concerning the changes:

> Context structures

Sounds like going the the correct correction, however this could result in some refactoring required in user's (ours :D) code.

> Reconsider default fallback field type
> […]
> This tends to confuse new users.
> […], this would then fail with an error message that would hopefully make it more clear what is happening.

That's also something we observed. Code touching `nanopb` is still a hurdle for some of our devs, so anything which makes it more user friendly helps. We all know that documentation is not always read and also not the biggest strength of `nanopb`.

Add a possibility to use`#define`s  in addition (!) to the current `.options` files is welcomed. We current use a lot of `.options` files which we share about multiple components to have a consistent behaviour, however sometimes there a special requirements for single use cases where overriding the values from the `.options` files could help. Ideally the values from the `.options` files whould be also a available in the `#define`s so you could do something like
```
#define MYMESSAGE_MYFIELD_MAXSIZE MIN(5, MYMESSAGE_MYFIELD_MAXSIZE_FROM_OPTIONS)
```

> Memory buffers as the primary I/O mechanism
> […]
> Overwhelming majority of nanopb users decode and encode to memory buffers.

This holds also true for us. We welcome the planned direction. :)

> If the buffer is larger than 1 byte, it acts as a cache to reduce number of I/O requests.

We also implemented something similar around `nanopb`, so having this supported natively would simplify things in our side. 👍

> Especially keeping support for AVR progmem storage is difficult.

No support for AVRs required from our side. In general I have the feeling that AVR usage is on the decreasing side and they still could use the current version.

PetteriAimonen, 2025-11-17:
Ok, I think I finally have some time to work on this.

I've started the changes on [dev_1.0](https://github.com/nanopb/nanopb/tree/dev_1.0) branch.
I aim to keep the tests passing during development, but naturally it shouldn't be used for anything important yet.

Here is a list I'll update to track progress:

* [x] Context structures
    * [X] Rename `pb_istream_t` and `pb_ostream_t` to `pb_decode_ctx_t` and `pb_encode_ctx_t`
    * [X] Adjust substream handling so that `ctx` pointer is passed unchanged to callbacks (#776)
    * [ ] Add test coverage for `ctx` pointer keeping its value
    * [ ] Make allocator part of the context (#775)
* [ ] Reconsider default fallback field type
* [ ] Improve bytes field pointer type
* [ ] Memory buffers as the primary I/O mechanism
* [ ] Possibly rework the field descriptor format
* [ ] API and ABI version compatiblity
    * [X] Add `PB_API_VERSION` for enabling compatibility macros
    * [ ] Change `PB_API_VERSION_DEFAULT` to 1.0 (currently 0.4.0 compatibility is enabled by default)
    * [ ] Refactor feature-enable macros so that default ABI is fully-functional and version-stable
    * [ ] Add test case for ABI compatibility between versions
* Other issues for 1.0: https://github.com/nanopb/nanopb/milestone/4

ncooke3, 2026-03-04:
> I think I prefer 3.
> 
> [@paulb777](https://github.com/paulb777) Do you think 3. would allow to get rid of the `spm_headers` symlink folder, which is currently blocking [#925](https://github.com/nanopb/nanopb/pull/925)? Or would it have to be 4.?

Hi @PetteriAimonen, jumping in for @paulb777 here. Option 4 would be preferable for Firebase's use in the Apple developer ecosystem. I mocked it up (diff: https://github.com/nanopb/nanopb/compare/dev_1.0...ncooke3:nanopb:ncooke3/dev_1.0) and verified it works for Swift Package Manager and CocoaPods (via our CocoaPods wrapper repo [fork](https://github.com/google/nanopb-podspec/compare/master...google:nanopb-podspec:nc/mockup-nanopb-refactor)). 

Additionally, I confirmed that option 4 would allow for the removal of the `spm_headers` symlink folder.