nanopb/nanopb #1068: -s options does not work?

xiongyu0523, 2025-03-07:
Hi, 

When I use protoc with plugin, i can generate code with no issue, then i want to specify file level options by passing --nanopb_opt, it does not work and prompt "Could not make proto path relative: mangling_names:M_STRIP_PACKAGE: No such file or directory"

basically this works

```
protoc --plugin=protoc-gen-nanopb --nanopb_out=. msg.proto
```
but this doesn't
```
protoc --plugin=protoc-gen-nanopb --nanopb_opt=-s mangling_names:M_STRIP_PACKAGE --nanopb_out=. msg.proto
```

when i was using nanopb_generator.py directly, i can pass -s OPTION:VALUE to that script with no issue

PetteriAimonen, 2025-03-07:
You probably need to quote it:

     protoc --plugin=protoc-gen-nanopb "--nanopb_opt=-s mangling_names:M_STRIP_PACKAGE" --nanopb_out=. msg.proto

xiongyu0523, 2025-03-07:
It works thank you.

closed 2025-03-07 by xiongyu0523