nanopb/nanopb #1071: Generated nanopb_pb2.py doesn't depend on protoc

charlesnicholson, 2025-03-12:
Hello, and thanks for the nanopb project! I've shipped multiple mass-produced IoT devices with it and it's truly excellent software. :)

I encountered an interesting issue- we upgraded our protoc package to the newly-released v30.0 from v28.x, and also upgraded our `protobuf` Python package from 5.28.x to 6.3.x, and started getting this warning:

```
UserWarning: Protobuf gencode version 5.28.3 is exactly one major version older than the runtime version 6.30.0 at nanopb.proto. Please update the gencode to avoid compatibility violations in the next runtime release.
```

I tracked the issue down to the nanopb generator python code, specifically `generator/proto/__init__.py`, which quietly compiles `nanopb.proto` to `nanopb_pb2.py` in the generator directory, so that it can be reused by the generator in future runs.

The existing code does take into account the timestamp of the `nanopb.py` file to see if `nanopb_pb2.py` is stale, but it doesn't take into account the timestamp (or version) of the `protoc` executable that it uses to compile `nanopb.py`.

So, when we upgraded our protobuf toolchain and package and did a full clean rebuild, we still got these warnings emitted from the generator for proto files that we don't manage.

We have to wrap `nanopb_generator.py` in our own launcher so that it forces our sandboxed `protoc` to the front of the path, so it was simple to just add a step that deletes `generator/proto/nanopb_pb2.py` and sets `NANOPB_PB_TEMP_DIR` before building, to force the generator to never emit a pb2 file in the generator directory. But, I thought it would be nice if nanopb itself could take a look and decide based on the protoc compiler if `nanopb_pb2.py` was stale.

Thanks for reading, and thanks again for nanopb!

charlesnicholson, 2025-03-12:
I guess another option could be to let users take control of compiling and providing `nanopb_pb2.py` to the generator via an environment variable or something; that way I could just explicitly hook it up as a normal proto in my build and understand when it's dirty etc. Then the generator wouldn't need any more complex logic; it would just say "oh, `NANOPB_PB2_PY` is set, I'll just use the file at that env var value's location, and if it doesn't exist I'll raise an error"

PetteriAimonen, 2025-03-12:
I'm not sure if there is any simple way to detect the version differences. Recent versions of protoc seem to put the protoc version into comment in the generated `_pb2.py`, but that would then have to be parsed out. Currently the build is automatically redone if the import fails, but not for warnings.

If you manage the generation externally, you can set `NANOPB_PB2_NO_REBUILD` to disable the internal logic. If needed, `PYTHONPATH` can be used to load `nanopb_pb2.py` from a custom location.

(Myself I just wish that python-protobuf had a way to just import a `.proto` without having to bother with running the generator.)

charlesnicholson, 2025-03-12:
Thanks for the quick response; `NANOPB_PB2_NO_REBUILD` plus `PYTHONPATH` munging sounds like a better solution than deleting it every time and forcing the generator to redo the work.

I agree regarding how to determine compatibility; doing anything more involved than timestamp checking seems fragile and fraught with peril.

closed 2025-03-12 by charlesnicholson