nanopb/nanopb #1072: Question about pb_release.

iabdalkader, 2025-03-14:
Hi, if I have something like the following protocol, do I pass `Msg` fields and `&msg` or `Msg_Resp` fields and `&msg_resp` to  `pb_release`? Or either way works?

```
message Msg_Resp {
    uint32 count = 1;
    repeated entries = 2;   // FT_POINTER
}

message Msg {
    MsgId msg_id = 2;
    oneof payload {
        Msg_Rsep1 resp1;
        Msg_Resp2 resp2;
        ...
    }
}
```
Apologies if this has already been documented somewhere, or if this not the right place to ask.

PetteriAimonen, 2025-03-14:
Either will work. `pb_release()` recurses into submessages.

While debugging, you can confirm that fields are released as `pb_release()` sets them to `NULL`.

iabdalkader, 2025-03-14:
> Either will work. `pb_release()` recurses into submessages.

Amazing! Thanks for confirming.

closed 2025-03-14 by iabdalkader